dkms-project/dkms #2: Backing up current initrd can fill up partitions

scaronni, 2016-05-24:
According to: https://github.com/dell-oss/dkms/blob/master/dkms#L267-L283

An old initrd is saved with the "old-dkms" suffix at every rebuild, to revert to the previous initrd in case of problems. This is fine, but it can lead to a full /boot partition after a few update, as old initrd images are never deleted (ref: https://bugzilla.redhat.com/show_bug.cgi?id=1251602).

What is the best approach here? Is it ok if I create a patch that deletes all "old-dkms" images prior to starting a new initird build? If a new initird build is triggered, this basically means that the user is updating the kernel or the module, and in both cases it ensures us that the "current boot" is fine and there is no need to revert.

What do you think?

Thanks,
--Simone

jared-dominguez, 2016-05-24:
Ideally old initrds for still installed kernels should be kept around. I'm fine accepting a cleanup patch that saves those.

nightsky30, 2017-04-01:
Affecting Debian/Ubuntu as well:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=717584
https://bugs.launchpad.net/ubuntu/+source/dkms/+bug/1515513
https://bugs.launchpad.net/ubuntu/+source/unattended-upgrades/+bug/1357093

jarnos, 2017-04-13:
https://github.com/dell/dkms/pull/22

jarnos, 2017-04-14:
Backup files could be cleared in update-initramfs's delete function, too.

yuzaipiaofei, 2017-04-17:
@scaronni Thanks your code work.

closed 2017-04-17 by superm1

murraybd, 2017-10-05:
This is actually incorrect as the remove_initrd_backup command runs before the dkms remove command which actually makes the .old-dkms file.

jarnos, 2017-10-05:
Why does it make a .old-dkms file when it runs dkms remove?

murraybd, 2017-10-05:
Because do_uninstall calls make_initrd and dkms, as far as I can tell, doesn't know the kernel is scheduled for removal just that the module is being removed so it remakes the initrd even though its getting removed.

jarnos, 2017-10-06:
Well, wouldn't it be better to tell make_initrd that kernel is being removed so there is no need to create a backup?