dkms-project/dkms #3: Add a not-so-quiet mode

Corax26, 2016-06-22:
dkms currently supports a flag to run in quiet mode, `-q`. The problem with this mode is that it is so quiet that it actually doesn't output anything at all, even errors are silenced. This is a problem for example in the dkms hooks in Arch Linux, where `-q` used to be used but got finally replaced with `> /dev/null` to avoid silencing errors (see [1], [2]).

I think that a less quiet mode would be more useful. I don't think it is expected of a program like dkms to silence error messages even in quiet mode, so the semantics of `-q` could be modified. Otherwise, a new option might be added, I think both options make sense.

[1] : https://bugs.archlinux.org/task/49080
[2] : https://git.archlinux.org/svntogit/packages.git/commit/trunk?h=packages/dkms&id=71ad07ba08fa92023219e54ba44867fd2833046d

yuzaipiaofei, 2017-05-23:
if you use a  quiet mode that means you do not need much debug information.
Then when you meet some problems . you can enable the debug parameter.
So  i would like to close this issue.
Thank you .

closed 2017-05-23 by yuzaipiaofei

Corax26, 2017-05-23:
I don't agree. Consider for instance `grep -q`: it outputs nothing on stdout, but still outputs basic diagnostic on stderr (non-existing files or invalid regex). It's the same thing here: if you use `-q` I agree you don't need *much* information, but that doesn't mean you don't want *any* information. Right now you can only tell there was an error by checking the exit code, and you have no idea about the nature of the error. At the very least, dkms should say that compilation failed (without necessarily the full error message).

Another very good example is xz, no need to go into that level of granularity, but that's mainly to illustrate that a single `-q` is not expected to silence errors:

>-q, --quiet
>    Suppress warnings and notices. Specify this twice to suppress errors too. This option has no effect on the exit status. That is, even if a warning was suppressed, the exit status to indicate a warning is still used.