dkms-project/dkms #12: DKMS does not detect installed modules if they are xz compressed

kaechele, 2016-11-20:
Fedora compresses it's kernel modules as xz since version 21 of the distribution.
This results in kernel modules having a .ko.xz suffix.
DKMS' set_module_suffix function does not handle this case. Therefor previously installed modules in the kernel tree are not found.
This problem surfaces when replacing a module that lives in Fedora's kernel-modules-extra package. Those kernel modules are installed into /extra and thus they override even their DKMS installed and updated version.

Example: batman-adv
Lives in the kernel-modules-extra package: /lib/modules/4.8.6-300.fc25.x86_64/extra/net/batman-adv/batman-adv.ko.xz
DKMS installs it's version as: /lib/modules/4.8.6-300.fc25.x86_64/extra/batman-adv.ko

kaechele, 2016-11-20:
Just noticed that there is already a pull request: https://github.com/dell/dkms/pull/4

closed 2016-11-20 by kaechele

superm1, 2016-11-21:
It looks to me that the person who was working on that pull request hasn't sorted out some of the requested fixes for a long time (August!).  If you want to take a stab at fixing the problems in their pull request and make a new one, please feel free to do so.

DocMAX, 2018-02-03:
Still with version 2.5 modules are created with .ko extension, not .ko.xz!