dkms-project/dkms #15: 'dkms remove' should probably not remove the module from the dkms_tree

ensonic, 2016-12-22:
This code https://github.com/dell/dkms/blob/master/dkms#L1849-L1858 has a problematic side effect.
When upgrading a kernel (I am using debian), the kernel packages prerm script (https://github.com/dell/dkms/blob/master/kernel_prerm.d_dkms#L12) will call 'dkms remove'. If this was the only installed kernel, all dkms modules are not only uninstalled and removed, but also completely purged. 
Then the new kernel is installed and the dkms autoinstaller won't find any modules to reinstall.

I see a few options:

1. only purge modules when calling 'dkms remove --all'
1. let dkms driver packages call a new 'dkms cleanup' step

Any thoughts?

yuzaipiaofei, 2017-03-11:
This issue has been fixed in:
https://github.com/yuzaipiaofei/dkms/commit/0c19129b5d1f8e03498f6f2455ad9f7e14e9e606

yuzaipiaofei, 2017-03-11:
Codes has been ready to merge to master now .
https://github.com/dell/dkms/pull/20

yuzaipiaofei, 2017-03-11:
@ensonic 
Could  you help to verify your issue with this patch ?

closed 2017-04-20 by superm1

ensonic, 2017-04-24:
We've upstreamed the patch and don't need the dkms anymore. I had worked around the issue, by manually managing a fake version (0.0) to keep the driver installed.