dkms-project/dkms #17: dkms module  rebuilt failed  when updating kernel version

yuzaipiaofei, 2017-01-18:
**When I updated kernel version , old dkms package will not be rebuilt ,just link old dkms modules to weak-update directory .  But I need to rebuild it for new kernel image .  I got several similar issues  reported in the past years .  so I need to record it here ,and find a workable solution to fix this .**
 

[root@localhost 3.10.0-514.el7.x86_64]# yum update kernel kernel-headers kernel-devel
\Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: mirrors.cn99.com
 * extras: mirrors.cn99.com
 * updates: mirrors.cn99.com
Resolving Dependencies
--> Running transaction check
---> Package kernel.x86_64 0:3.10.0-514.2.2.el7 will be installed
---> Package kernel-devel.x86_64 0:3.10.0-514.2.2.el7 will be installed
---> Package kernel-headers.x86_64 0:3.10.0-514.el7 will be updated
---> Package kernel-headers.x86_64 0:3.10.0-514.2.2.el7 will be an update
--> Finished Dependency Resolution

Dependencies Resolved

============================================================================================================================================
 Package                             Arch                        Version                                 Repository                    Size
============================================================================================================================================
Installing:
 kernel                              x86_64                      3.10.0-514.2.2.el7                      updates                       37 M
 kernel-devel                        x86_64                      3.10.0-514.2.2.el7                      updates                       13 M
Updating:
 kernel-headers                      x86_64                      3.10.0-514.2.2.el7                      updates                      4.8 M

Transaction Summary
============================================================================================================================================
Install  2 Packages
Upgrade  1 Package

Total size: 55 M
Is this ok [y/d/N]: y
Is this ok [y/d/N]: yt
Is this ok [y/d/N]: y
Downloading packages:
Running transaction check
Running transaction test
Transaction test succeeded
Running transaction
  Installing : kernel-devel-3.10.0-514.2.2.el7.x86_64                                                                                   1/4 
  Installing : kernel-3.10.0-514.2.2.el7.x86_64                                                                                         2/4 
  Updating   : kernel-headers-3.10.0-514.2.2.el7.x86_64                                                                                 3/4 
  Cleanup    : kernel-headers-3.10.0-514.el7.x86_64                                                                                     4/4 

Good news! Module version 367.44 for nvidia.ko
exactly matches what is already found in kernel 3.10.0-514.2.2.el7.x86_64.
DKMS will not replace this module.
You may override by specifying --force.
Error! Module version 6BE4B82009EDF06D87F035E for nvidia-uvm.ko
is not newer than what is already found in kernel 3.10.0-514.2.2.el7.x86_64 (367.44).
You may override by specifying --force.

Good news! Module version 367.44 for nvidia-modeset.ko
exactly matches what is already found in kernel 3.10.0-514.2.2.el7.x86_64.
DKMS will not replace this module.
You may override by specifying --force.

Good news! Module version 367.44 for nvidia-drm.ko
exactly matches what is already found in kernel 3.10.0-514.2.2.el7.x86_64.
DKMS will not replace this module.
You may override by specifying --force.
modinfo: ERROR: Module /lib/modules/3.10.0-514.el7.x86_64/weak-updates/ not found.
modinfo: ERROR: Module /lib/modules/3.10.0-514.2.2.el7.x86_64/nvidia.ko not found.
modprobe: FATAL: Module /lib/modules/3.10.0-514.2.2.el7.x86_64/nvidia.ko not found.
Warning: Module nvidia.ko from kernel  has no modversions, so it cannot be reused for kernel 3.10.0-514.el7.x86_64
modinfo: ERROR: Module /lib/modules/3.10.0-514.el7.x86_64/weak-updates/ not found.
modinfo: ERROR: Module /lib/modules/3.10.0-514.2.2.el7.x86_64/nvidia-uvm.ko not found.
modprobe: FATAL: Module /lib/modules/3.10.0-514.2.2.el7.x86_64/nvidia-uvm.ko not found.
Warning: Module nvidia-uvm.ko from kernel  has no modversions, so it cannot be reused for kernel 3.10.0-514.el7.x86_64
modinfo: ERROR: Module /lib/modules/3.10.0-514.el7.x86_64/weak-updates/ not found.
modinfo: ERROR: Module /lib/modules/3.10.0-514.2.2.el7.x86_64/nvidia-modeset.ko not found.
modprobe: FATAL: Module /lib/modules/3.10.0-514.2.2.el7.x86_64/nvidia-modeset.ko not found.
Warning: Module nvidia-modeset.ko from kernel  has no modversions, so it cannot be reused for kernel 3.10.0-514.el7.x86_64
modinfo: ERROR: Module /lib/modules/3.10.0-514.el7.x86_64/weak-updates/ not found.
modinfo: ERROR: Module /lib/modules/3.10.0-514.2.2.el7.x86_64/nvidia-drm.ko not found.
modprobe: FATAL: Module /lib/modules/3.10.0-514.2.2.el7.x86_64/nvidia-drm.ko not found.
Warning: Module nvidia-drm.ko from kernel  has no modversions, so it cannot be reused for kernel 3.10.0-514.el7.x86_64
  Verifying  : kernel-headers-3.10.0-514.2.2.el7.x86_64                                                                                 1/4 
  Verifying  : kernel-3.10.0-514.2.2.el7.x86_64                                                                                         2/4 
  Verifying  : kernel-devel-3.10.0-514.2.2.el7.x86_64                                                                                   3/4 
  Verifying  : kernel-headers-3.10.0-514.el7.x86_64                                                                                     4/4 

Installed:
  kernel.x86_64 0:3.10.0-514.2.2.el7                                kernel-devel.x86_64 0:3.10.0-514.2.2.el7                               

Updated:
  kernel-headers.x86_64 0:3.10.0-514.2.2.el7                                                                                                

Complete!

-------------------

[root@localhost 3.10.0-514.el7.x86_64]# dkms status
nvidia, 367.44, 3.10.0-514.2.2.el7.x86_64, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
nvidia, 367.44, 3.10.0-514.el7.x86_64, x86_64: installed
[root@localhost 3.10.0-514.el7.x86_64]# 

Similar issue reported : 
https://github.com/zfsonlinux/zfs/issues/3801
https://www.centos.org/forums/viewtopic.php?t=56627
http://lists.us.dell.com/pipermail/dkms-devel/2007-May/000562.html

yuzaipiaofei, 2017-02-19:
If I update  kernel with  rpm -Uvvh *.rpm  ,including (kernel-3.10.0-514.9.1.el7.x86_64.rpm  kernel-devel-3.10.0-514.9.1.el7.x86_64.rpm  kernel-headers-3.10.0-514.9.1.el7.x86_64.rpm)
Then i will get below errors , because new kernel update will remove old dkms module and rebuild for new kernel ,but current dkms building work will select old kernel source code to build and failed.
get logs from make.log showing below errors :
DKMS make.log for openvswitch-1.4 for kernel **3.10.0-514.9.1.el7.x86_64** (x86_64)
Sat Feb 18 20:51:55 PST 2017
make -C /lib/modules/3.10.0-514.6.1.el7.x86_64/build M=/var/lib/dkms/openvswitch/1.4/build modules
make: *** /lib/modules/3.10.0-514.6.1.el7.x86_64/build: No such file or directory.  Stop.
make: *** [all] Error 2

yuzaipiaofei, 2017-02-19:
Dkms service did not build dkms modules for new kernel source codes. 
If adding dkms modules manually,and enable dkms service ,then reboot .
I found that modules built correctly and installed . 
Something wrong for RHEL 7.3 OS version, need to do more debug and find root cause .

yuzaipiaofei, 2017-03-11:
Has been fixed in the commit :
https://github.com/yuzaipiaofei/dkms/commit/0c19129b5d1f8e03498f6f2455ad9f7e14e9e606

It will be merged in master.

closed 2017-03-11 by yuzaipiaofei

brian-lamb-software-engineer, 2018-02-19:
This issue is ongoing i think, and reoccurs often I believe.  So, whats the way to get the modules working again?

I got the same errors, 
`ERROR *.ko not found`
```
$ /sbin/modprobe zfs
modprobe: FATAL: Module zfs not found.
```
```
# dkms remove -m zfs -v 0.7.6 --all
# dkms remove -m spl -v 0.7.6 --all
# dkms add -m spl -v 0.7.6
# dkms add -m zfs -v 0.7.6
# dkms install -m spl -v 0.7.6
# dkms install -m zfs -v 0.7.6
```

```
$ /sbin/modprobe zfs
modprobe: FATAL: Module zfs not found.
```

```
 dkms status
spl, 0.7.6, 3.10.0-693.11.1.el7.x86_64, x86_64: installed (WARNING! Diff between built and installed module!)
zfs, 0.7.6, 3.10.0-693.11.1.el7.x86_64, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
```
Then I see this https://github.com/zfsonlinux/zfs/issues/1155 talking about reinstalling linux headers, but i cant figure out how to do that, I tried yum list, list with wildcard, I just cant find the right package, so dont want to risk anything. 

Where from here?

brian-lamb-software-engineer, 2018-03-20:
Since that day, the zpool will no longer automount, infact it gets locked, i have to export it before i import it, probably because it only gets half imported on boot.  

So, now that I see that a yum update has updated ZFS again, I want to try to undo this, but im not sure if this is the correct way.  

$ dkms status
spl, 0.7.7, 3.10.0-693.11.1.el7.x86_64, x86_64: installed
zfs, 0.7.7, 3.10.0-693.11.1.el7.x86_64, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)

# dkms remove -m zfs -v 0.7.7 --all
# dkms remove -m spl -v 0.7.7 --all
# dkms add -m spl 
# dkms add -m zfs 
# dkms install -m spl
# dkms install -m zfs

Would anybody know how i back myself out of this to see if it will work by itself now?

mailinglists35, 2018-11-15:
I'm also getting the fstatat error when upgrading 0.8.0-rc1 to rc2

https://pastebin.com/UVrbPX4A