dkms-project/dkms #18: When there's only one version of a module, allow to omit the module version.

khfeng, 2017-01-20:
As someone who regularly remove/install dkms, I found specifying the module version is redundant in most cases.

I'd like to use `dkms install <module>` instead of `dkms install <module>/<version>`, if there's only one version of the module.

If this new behavior is acceptable and does not break scripts that already depend on DKMS, I can make a patch for this.

yuzaipiaofei, 2017-01-21:
For someone want to use  dkms install frequently ,I suggest you write your own script to do this .
debug script  is more helpful than change original dkms install command .

khfeng, 2017-01-21:
I need this because I modify module source directly in `/usr/src` for debugging, so recompile modules is necessary in my case.
I don't see debug script is helpful in this case.

yuzaipiaofei, 2017-02-19:
In most situation ,Linux user have some dkms modules installed ,I do not think that we need to change function codes for this debuging or convenience purpose .
@jared-dominguez 
 What is your opinion on this request ?

khfeng, 2017-02-21:
I don't think I fully understand what your concern is, but at least this kind of change won't break existing functionality.

The old behaviour (version explicitly specified) will keep working.

yuzaipiaofei, 2018-01-25:
@khfeng 
Hi Kai heng.
Normally, we can choose the module with quick TAB key.
Then if you do not have any questions. 
I will close the issue after one week.

khfeng, 2018-01-26:
I don't think you really understand what I am asking for.

To simply put: why do we need to specify version number, when there's _only_ one version available?

jared-dominguez, 2018-01-26:
@khfeng,

If I'm understanding right, this is entirely different from, say, <https://github.com/dell/dkms/pull/5>. I think the following should be acceptable:

1. User types 'dkms install <module>[/<version>]'
2. dkms checks /usr/src for <module> instances
3a. If <version> is specified on command line, note that version and continue to step 4. If no version is specified and a one version flag is specified, continue to step 3b.
3b. If *only* one version is in /usr/src and a module of the same name is not already installed, assume version is same as found in /usr/src and note that version number.
4. Use the version number as noted.

Because this is a behavioral change, I think we should only enable this with a flag ("-1" perhaps?). That is, I think it's okay to do this as an _optional_ change. I don't want this to devolve into a semantic change in existing functionality like <https://github.com/dell/dkms/pull/5> would be, as a semantic change could result in unintended consequences for distributions depending on DKMS.

yuzaipiaofei, 2019-05-13:
@khfeng 
Hi  Feng.
Could you help to make the patch for this improvement?