dkms-project/dkms #19: dkms.service systemd run target issue (vmware modules/vmware.service)

feuler, 2017-02-09:
Hi.

i've managed to build a vmware modules dkms rpm for SUSE (See: https://build.opensuse.org/package/show/home:feuler:vmware-modules-12/vmware-modules-12)

But dkms.service here and on software.opensuse.com have "[Install] WantedBy=multi-user.target" and  opensuse sources also have "[Unit] Before=display-manager.service".

Problem is that vmware workstation 12.5  /etc/init.d/vmware service is running parallel with dkms build (because of vmnet dhcp service) and won't start because module build is not finished. 
After boot is complete you can restart the failed vmware service without a problem.

Question is: Is there any reason for the systemd target in dkms.service ? Can i change service file safely to ??:
[Unit] section:
Before=network-pre.target
Wants=network-pre.target
[Install]
WantedBy=network-pre.target

And what is the first safe run target in systemd for dkms ?

Best regards

feuler, 2017-02-17:
Update:
This week i've built another dkms module (intel igb kernel driver update fix).
And after kernel upgrade + reboot with dkms bulding vmware-modules and intel igb driver, there was only a "lo" network interface. Somehow xorg+gdm was started while dkms modules where still compiling. That seemed to have triggered NetworkManager/network service getting started.
So, i also had to add "Before=graphical.target" to the dkms service so that networking gets initialized correctly after kernel upgrade reboot.

[Unit]
Before=network-pre.target graphical.target
Wants=network-pre.target

yuzaipiaofei, 2017-02-19:
HI feuler:
Could you tell me which Suse release version you find this issue ?

feuler, 2017-02-22:
Suse SLED 12 SP2 (and SP1)

feuler, 2017-02-22:
And these are the two dkms modules vmware/intel igb:
[http://feuler.bplaced.net/vmware-modules-12-1.0.src.rpm](url)
[http://feuler.bplaced.net/vmware-modules-12-1.0.x86_64.rpm](url)
[http://feuler.bplaced.net/igb-dkms-5.3.5.4-1.0.src.rpm](url)
[http://feuler.bplaced.net/igb-dkms-5.3.5.4-1.0.x86_64.rpm](url)

closed 2022-05-25 by scaronni