dkms-project/dkms #26: Can't remove dkms on centos 6

johnlongo, 2017-06-19:
Can't remove dkms on centos 6.

see following:

Downloading Packages:
Running rpm_check_debug
Running Transaction Test
Transaction Test Succeeded
Running Transaction
/var/tmp/rpm-tmp.fXOK7F: line 3: fg: no job control
error: %preun(dkms-2.2.0.3-30.1.noarch) scriptlet failed, exit status 1
Error in PREUN scriptlet in rpm package dkms
dkms-2.2.0.3-30.1.noarch was supposed to be removed but is not!
  Verifying  : dkms-2.2.0.3-30.1.noarch                                                                        1/1 

Failed:
  dkms.noarch 0:2.2.0.3-30.1                                                                                       

Complete!
[root@applied-db2prod-1 ~]#

yuzaipiaofei, 2017-07-07:
seems that your package encounter broken issue,if you delete or change default dkms files. maybe lead you failed to uninstall original dkms package. 
you can have a try to use this command.
#rpm -evvf dkms 

error: %preun(dkms-2.2.0.3-30.1.noarch) scriptlet failed, exit status 1

scaronni, 2018-01-24:
You can use `rpm -e --noscripts dkms` and it should be removed. Also, the package is officially mantained in EPEL for CentOS/RHEL, so please use the version there.

Thanks.

closed 2018-01-24 by scaronni