dkms-project/dkms #29: Creating optional modules

jasmin-j, 2017-07-23:
Hi!

I try to make a DKMS for the linux media drivers.
The problem is, that the media build systems builds the modules depending on the Kernel version.
If I specify all possible modules in the dkms.conf file, the build will fail if the media build system does not built all the modules do to the currently built Kernel version and modules not supported for it.

Do you see any chance how I can realize a DKMS supporting this?
I thought to generate the dkms.conf dynamically depending on the Kernel version, but this requires a kind of DKMS chaining because DKMS reads the configuration only once.

Any further help is very welcome.

BR,
Jasmin

superm1, 2017-07-25:
@jasmin-j I think you should be able to use shell scripting directly in the dkms.conf.  Have you already tried that?

jasmin-j, 2017-07-26:
No, thank you for the hint!
May I ask how I can do that and where to find a documentation or better an example?
BR,
Jasmin

superm1, 2017-07-28:
If you don't see it in the man page it doesn't exist.  I know i've seen packages do this in the past, but I don't recall which ones.

closed 2017-07-28 by superm1

jasmin-j, 2017-09-09:
Hi!

I was able to generate a media-build-dkms package.
I used a POST_BUILD script to dynamically create nearly the whole content of the dkms.conf file with all the modules built.
If someone is interested how this can be done, look to:
https://github.com/jasmin-j/media_build/tree/dkms-master
-> dkms.conf
-> gen_dkms_dyn_conf.sh
-> Makefile.dkms

BR,
Jasmin