dkms-project/dkms #31: install/uninstall inconsistent

jasmin-j, 2017-09-09:
Hi!

While testing my new media-build-dkms (https://github.com/jasmin-j/media_build/tree/dkms-master), I discovered several issues concerning the install/uninstall commands.

First of all dkms uses the version got from command modinfo to decide, if a newly built module is newer or equal than the existing one. I am not sure if this is really sufficient.
In my case there are several drivers which didn't increase the version, but the underlying media-core has been changed heavily. So they haven't been installed, although they should. When I want to solve this, I would need "--force". But the "dkms_common.postinst" has no possibility to add "--force" by e.g. an environment variable.
Even more, "template-dkms-mkdeb/debian/postinst", prefer the system wide installed "common.postinst" script. So there is no possibility for a package maintainer to add the "--force" option, other than writing his own postinst script.

I would suggest to add a possibility to add options to the dkms commands used in "common.postinst" preferably by environment variables, because they can be easily set in the package "debian/postinst" file or even at the command line of the "apt-get ..." command.

When it comes to package removing, dkms checks if the installed module and the one from the dkms module directory (/var/lib/dkms/<module>/<version>/<kernel>/<arch>/module) are the same. In case when the module was not installed, "find_actual_dest_module_location" will return an empty string which will let variable "real_dest_module_location" empty, which will then lead to an error of the "rm -f ..." command.

I see here some problems.
a) It is not a good idea to check the module version when installing, but the whole file when uninstalling. This is inconsistent. I don't know if it would be possible to diff the modules at the install step also, instead of checking the version. This would also solve the problem mentioned in the beginning. On the other hand, the modules might be always different, if there is a date or time stored in the module .ko file, which I don't know.

Another possibility would be to check in the uninstall function also the versions from modinfo, but this would not uninstall the module, if it was installed with "--force". So this is no solution.

b) When "real_dest_module_location" is empty, no module was installed, so it would be the best to simply do nothing (doing a continue).

BR,
Jasmin

jasmin-j, 2017-09-10:
In the beginning of the previous message, I wrote about "--force".
I just found the possibility to force the installation by a file "/usr/share/dkms/modules_to_force_install/<dkms-module>". I will try that to overcome the initial problem with the not updated "*.ko" module versions.

xuzhen, 2023-01-25:
Should be fixed by #275

closed 2023-01-25 by xuzhen