dkms-project/dkms #33: DKMS is POSIXly incorrect - breaks package installs

jelmd, 2017-09-18:
If one has set its env var POSIXLY_CORRECT (which is default for all our users and admins), DKMS fails and prevents proper kmod compilation/installation and screws up installation of packages, which depend on in (direct or indirect). E.g.:
```
# apt-get install --no-install-recommends nvidia-384 nvidia-settings libcuda1-384 nvidia-opencl-icd-384 fakeroot
Reading package lists... Done
...
Loading new nvidia-384-384.66 DKMS files...
/usr/sbin/dkms: line 193: syntax error near unexpected token `<'
/usr/sbin/dkms: line 193: `	done < <(module_status_built "$module" "$module_version")'
dpkg: error processing package nvidia-384 (--configure):
 subprocess installed post-installation script returned error exit status 2
dpkg: dependency problems prevent configuration of libcuda1-384:
 libcuda1-384 depends on nvidia-384 (>= 384.66); however:
  Package nvidia-384 is not configured yet.
...
```

I'm not sure, why such an obscure syntax like `while read line; do : ; done < <($cmd)` is used.
IMHO all such occurences should be replaced by the POSIX compatible `($cmd) | while read line; do : ; done` syntax.

yuzaipiaofei, 2017-09-26:
could you provide the dkms package which version you are using?
#dkms --version
#dpkg -l | grep dkms

jelmd, 2017-09-26:
Yepp (it is Ubuntu xenial aka 16.04):
``` 
> dkms --version
dkms: 2.2.0.3
 > dpkg -l | grep dkms
ii  dkms                             2.2.0.3-2ubuntu11.3                        all          Dynamic Kernel Module Support Framework
```

yuzaipiaofei, 2017-11-30:
@jelmd 
The dkms version you use is Ubuntu version.
I suggest that you build the dkms package by you self and have a try again.
The latest version is v2.4 here.

jelmd, 2017-11-30:
@yuzaipiaofei Thanx for the info! It is a little bit tricky: The dkms install gets triggered by nvidia (and perhaps other packages), but actually it is not really needed, because this is in an LXC container, where dkms is useless (for compiling stuff I wrote a dkms replacement, which is more efficient, straight forward and flexible wrt. paths to use, does not require to be root, etc.). Unfortunately Ubuntu/Debian ancient package stuff is still lxc unaware, which led us to the plan to create our own pkg repo and put an empty dkms package there to just to satisfy the dependencies ...

yuzaipiaofei, 2017-12-22:
@jelmd  if you are going to install NVIDIA Driver,then you can just install official Release of NVIDIA .it will not require you to use dkms.

Official NVIDIA driver it is kind of xxx.run package. if you install NVIDIA from deb package. then you will use dkms for the driver.

superm1, 2018-01-02:
I wonder if the proper solution is to change DKMS behavior to not run in containers (but allow a new command line override if desired; such as for CI or something).

Here's some advice for checking if you're running in a container:
https://stackoverflow.com/questions/20010199/determining-if-a-process-runs-inside-lxc-docker

jelmd, 2018-01-02:
@yuzaipiaofei Hmmm, but I guess than I would miss all the ubuntu specific tweaks?

Anyway, it is getting OT - no matter how the nvidia problem gets solved or whether dkms gets "container aware", the dkms remains POSIXLY incorrect.

superm1, 2018-01-02:
IIRC NVIDIA actually detects if DKMS is on the system from their xxx.run and uses it if available too.  So this may also be affected in that situation.

@jelmd would you mind submitting a PR with what you recommended for fixing the ```POSIXLY_CORRECT```?  I think that's step 1.
Step 2 DKMS should get container aware.

jelmd, 2018-01-02:
@superm1 : yes, I can do it. Would do it as suggested, i.e. `($cmd) | while read line; do : ; done`. Not sure, whether it is sufficient because of the annoying last-pipe-in-a-subshell bashism ...

martingalvan, 2018-11-20:
Is this still lingering? Sounds like a critical usability bug.

E5ten, 2019-10-01:
The suggested solution wouldn't fix the issue, read is also used directly that way to set variables that are used later, and piping into read in the same style as the suggested solution would set those variables in a subshell and not work.
e.g. `read -a kernels_module < <(find_module "$lib_tree" "${4}")`

TriMoon, 2020-06-08:
Maybe it's an idea to make use of FIFO's?
That way the output of `$cmd` can be redirected to the FIFO and the `read` can use the FIFO to read...
[Some info and example wrt FIFO in bash](https://www.linuxjournal.com/content/using-named-pipes-fifos-bash)...

scaronni, 2025-01-28:
Please check the latest version, the code has been refactored many times since this bug was opened. In the unlikely event the issue is still there, please open another bug. Thanks.

closed 2025-01-28 by scaronni