dkms-project/dkms #38: Compressed modules (*.ko.gz)

DocMAX, 2017-11-18:
I installed a patch for alx.ko module and i noticed the original module is alx.ko.gz, but DKMS put alx.ko into /lib/modules/4.13.12-1-ARCH/kernel/drivers/net/ethernet/atheros/alx/ thus kernel is loading the wrong module!

How can i make DKMS create and replace the alx.ko.gz module?

d3matt, 2017-11-27:
I'm seeing something similar on RHEL7. With kernel versions kernel-3.10.0-693.el7.x86_64 and kernel-3.10.0-693.2.2.el7.x86_64 they're using xz compressed modules...

yuzaipiaofei, 2017-11-30:
@d3matt 
Thanks for the feedback and the  patch.
@DocMAX 
Can you help to verify if the patch can solve your problem?
If the fix works, i will merge the patch soon.

closed 2018-01-24 by scaronni

DocMAX, 2018-02-03:
Still doesnt work. It's not creating alx.ko.xz but alx.ko instead. Arch linux Kernel 4.14

yuzaipiaofei, 2018-04-25:
HI @DocMAX 
new release come and new version can support xz compress format.
if you have no question ,i will close the issue soon.

[dell@localhost extra]$ file rtbth.ko.xz
rtbth.ko.xz: XZ compressed data

[dell@localhost extra]$ modinfo rtbth
filename: /lib/modules/4.15.17-200.fc26.x86_64/extra/rtbth.ko.xz
version: 3.9.5
license: GPL
description: Support for Ralink Bluetooth RT3290 Cards
author: Ralink Tech.
srcversion: E3E18A59C0E0B1FFBA15FE1
alias: pci:v00001814d00003298svsdbcsci*
depends: bluetooth
retpoline: Y
name: rtbth
vermagic: 4.15.17-200.fc26.x86_64 SMP mod_unload

[dell@localhost extra]$ dkms status
rtbth, 1.0, 4.15.17-200.fc26.x86_64, x86_64: installed

[dell@localhost extra]$ modprobe rtbth

[dell@localhost extra]$ lsmod | grep rtbth
rtbth 98304 0
bluetooth 593920 1 rtbth

PR #39 (2018-01-24): Compressed module support
Should resolve #38