dkms-project/dkms #41: Version and Date not up to date in manpage

mseeber, 2017-12-10:
I noticed that the date and version number in the manpage dkms.8 are not updated with the release version of dkms.

Is there sense in inserting the current date/version in the tarball/install targets in the Makefile? If so I could provide a patch later.

superm1, 2017-12-11:
I think that's a great idea at least with the release version.  I'd argue the date should just be removed though.  It doesn't really offer any value.

scaronni, 2018-01-24:
Patch here, if that is ok for you I can merge it: https://github.com/dell/dkms/pull/45

yuzaipiaofei, 2018-01-25:
@scaronni 
As Mario said .the date dose not have much value change.
But i merge your patch now.  thanks for the improvement work.
I would like to see if the patch can work for the next release.

yuzaipiaofei, 2018-01-25:
If you do not have new issue for this .
I would close this issue now.

closed 2018-01-25 by yuzaipiaofei