dkms-project/dkms #42: Is there a way to skip version check from dkms.conf?

bp0, 2018-01-01:
Is there some way to make auto install always --force from dkms.conf? Or perhaps, some way to skip the version check? If the module is using the git hash as its module version, then it is not comparable in any regular way.
I tried to use PRE_INSTALL="true", as the man page said that PRE_INSTALL could be used for a custom version check, and that did not work. Also, I tried PRE_INSTALL="dkms install --force -m ..." in an attempt to just hack around the version check, but it didn't work either.

This is the attempted use of DKMS that fails when the kernel package is updated, for reference: https://github.com/groeck/it87/commit/6c1e48b4f16b65b6e2ff95d950b250581b0cbe91

yuzaipiaofei, 2018-01-25:
The  --force option to be used when the version checking failed.
I do not think we can always need to install packages with force .
@bp0

bp0, 2018-01-26:
DKMS' version check appears to always fail if the module doesn't provide a module version. For it87, the solution was simply to add a MODULE_VERSION() to the source (https://github.com/groeck/it87/commit/5edd7bb6f908b66b3a2a0d0a7ec9e04d723147e5). To me it still seems like there should be a way to skip the version check, or override it, via dkms.conf.

yuzaipiaofei, 2019-05-13:
@bp0 ,I think the module missing is the problem of the module, it is not the right way to ignore the normal check process for dkms .
The issue will  be close if you have no further questions about this.

alex-tu-cc, 2019-09-23:
a related issue that becuase of that weird version comparing rule to compare the git hash that some released dkms add "--force" by default. Then obsoleted_by parameter in dkms.conf will be always be overrided. So, I commited a patch there : https://github.com/dell/dkms/issues/89 , to make obsoleted_by has higher priority than "--force"

But to think a bit more, it might impact people who really want to use "--force" to override everything blindly. So, I'm thinking to remove the part of comparing git hash which looks not really make sense. 

So, the proposal will be: if the module version is not defined, then install that dkms without comparing git hash. Say, people must define a module version if they would like to utilize that version comparing. After that, we can carry back "--force" to own the highest priority to override everything blindly and obsoleted_by will not be impacted anymore.

How do you think?

alex-tu-cc, 2019-09-23:
after discussing with some other engineers, it seems better to preserve current version checking logic because we don't know who might count on it to build their logic on their system.

So, the new proposal will be: to provide a new parameter "--force-version-override" for those people who want to install dkms without version checking, but still want "obsoleted_by in dkms.conf" to retire the dkms after the fix already in newer kernel.

How do you think?

bviktor, 2019-12-06:
This is a problem when the mainline version of a driver diverges from upstream, like the Intel network drivers.

https://github.com/noobient/noobuntu/issues/27

5.3.5.42 is newer than 5.4.0-k found in the kernel, and you _can_ override this during `dkms install`. The problem arises when the DKMS driver is already installed, and you receive a newer kernel. It rebuilds all the modules, but throws away these DKMS drivers with smaller version numbers, and apparently there's no way to circumvent this currently.

alex-tu-cc, 2020-01-01:
> This is a problem when the mainline version of a driver diverges from upstream, like the Intel network drivers.
> 
> [noobient/noobuntu#27](https://github.com/noobient/noobuntu/issues/27)
> 
> 5.3.5.42 is newer than 5.4.0-k found in the kernel, and you _can_ override this during `dkms install`. The problem arises when the DKMS driver is already installed, and you receive a newer kernel. It rebuilds all the modules, but throws away these DKMS drivers with smaller version numbers, and apparently there's no way to circumvent this currently.

I thought it be covered by https://github.com/dell/dkms/pull/98
Can you please add "set -x" in the head of /usr/sbin/dkms , then install a new kernel and redirect message to a file and attache it here?

I will expect every time building dkms will check force-version-override.

bviktor, 2020-01-06:
Uh, that might actually work, if Ubuntu actually supported it. Unsurprisingly [they don't](https://bugs.launchpad.net/ubuntu/+source/dkms/+bug/1845467). This patch is pretty recent so I'm not sure it will make it into 20.04 even, currently it sits on v2.7.1... but I'll keep an eye out. Until then, if someone finds a way that works on 18.04, please do let me know, thanks a lot.

closed 2022-05-25 by scaronni