dkms-project/dkms #48: Calculate value from within dkms.conf

ev-mp, 2018-02-19:
Hello all,
I'm trying to build debian dkms for several in-tree modules that require patches.
I'm looking to a way to do the following in DKMS.conf : 
> SELECTED_KERNEL=**$kernelver | cut -d- -f1**

In order to use it later as
> BUILT_MODULE_LOCATION[0]="$SELECTED_KERNEL/drivers/media/..."

So for instance with `$kernelver=4.4.0-112_generic` I would get `SELECTED_KERNEL=4.4.0`
I tried to change the DKMS.conf to bash script by adding '#!/bin/sh' and `chmod`ing accordingly, but still, the new variable is not being calculated.

Any comments will be helpful

ev-mp, 2018-02-19:
One more thing, according to [DKMS manual](https://manpages.debian.org/stretch/dkms/dkms.8.en.html) the file DKMS.conf is a shell-script:

> Note that the dkms.conf is really only a shell-script of variable definitions which are then sourced in by the dkms executable 

So at least in theory there are provisions for this trick to work

superm1, 2018-02-19:
> SELECTED_KERNEL=$kernelver | cut -d- -f1

I'd think you want it something like this:

`SELECTED_KERENL=$(echo $kernelver | cut -d- -f1)
`

ev-mp, 2018-02-20:
Thanks for suggestion, I was interrupted to another task, but will return, check and update.

ev-mp, 2018-02-27:
Hello, I want to update and confirm that the above syntax works! 
thanks for the great advice - it just saved me a copy-paste of three sections

closed 2018-02-27 by ev-mp