dkms-project/dkms #51: Next version release?

jansohn, 2018-04-21:
When is the next version release planned? I'm particularly interested in #38 ...

yuzaipiaofei, 2018-04-21:
HI @jansohn ：
I  will try to  test the patches soon and add the xz support .
Then the new version will be released.

yuzaipiaofei, 2018-04-21:
Close this .
Please feel free to reopen if you have any concern.

closed 2018-04-21 by yuzaipiaofei

yuzaipiaofei, 2018-04-25:
@jansohn 
new release come and support xz compress format.
[dell@localhost extra]$ file rtbth.ko.xz 
rtbth.ko.xz: XZ compressed data
[dell@localhost extra]$ modinfo rtbth 
filename:       /lib/modules/4.15.17-200.fc26.x86_64/extra/rtbth.ko.xz
version:        3.9.5
license:        GPL
description:    Support for Ralink Bluetooth RT3290 Cards
author:         Ralink Tech.
srcversion:     E3E18A59C0E0B1FFBA15FE1
alias:          pci:v00001814d00003298sv*sd*bc*sc*i*
depends:        bluetooth
retpoline:      Y
name:           rtbth
vermagic:       4.15.17-200.fc26.x86_64 SMP mod_unload 
[dell@localhost extra]$ dkms status 
rtbth, 1.0, 4.15.17-200.fc26.x86_64, x86_64: installed

[dell@localhost extra]$ modprobe  rtbth
[dell@localhost extra]$ lsmod | grep rtbth
rtbth                  98304  0
bluetooth             593920  1 rtbth