dkms-project/dkms #53: dkms mkdeb: post_install scripts are always non-executable

martingalvan, 2018-05-04:
Every time I try to use a POST_INSTALL script, I get the following:

`Warning: The post_install script is not executable.`

This is because of [this](https://github.com/dell/dkms/blob/master/template-dkms-mkdeb/Makefile#L15).

plouj, 2018-10-03:
The error would be more helpful if it included the value of ```$run```. Simply placing the ```post_install``` script into the correct sub-directory in ```/var/lib/dkms/$module/$module_version/source``` or just ```/usr/src/$module``` fixes the problem for me.

martingalvan, 2019-03-25:
Almost a year later and this issue still exists.

WowVeryLogin, 2021-01-29:
Yeap, having the same issue. Executable bit is just droped from the script file.

martingalvan, 2021-01-29:
Almost three years later and the issue is still there.

scaronni, 2021-01-30:
Anyone with a simple patch I can merge? I'm not a Debian user, but I can help. Pretty sure that in 3 years people have applied workarounds that we can reuse here.
A patch would close #53 and #55.

scaronni, 2021-10-30:
Distribution specific packaging was severely broken for most distributions and has been dropped in the latest release.

closed 2021-10-30 by scaronni

gourryinverse, 2021-11-03:
@scaronni can you tell us what we're supposed to be doing instead?

Bassem-Ramzy, 2022-12-18:
For the record, it runs with me now on `dkms 2.8`, `Ubuntu 20.04.1 LTS, 5.15.0-56-generic`. 
I get the same error when I set `POST_INSTALL="myscript.sh arg1"`, and worked when set to `POST_INSTALL="myscript.sh"` (without spaces and arguments)

farzadpanahi, 2023-06-14:
I think I know where the root cause of this issue is. 

### Problem

`dkms mkdeb` is using `template-dkms-mkdeb` template files. You can see that at [this line](https://github.com/dell/dkms/blob/4d476115049f6bf698dfd64acdaf45d612888914/template-dkms-mkdeb/Makefile#L15) permissions are set to `644` for all files. This will remove `x` permission for all files including script files:

```bash
install -d "$(SRC)"
cp -a $(NAME)-$(VERSION) $(SRC)
chmod 644 -R "$(SRC)/$(NAME)-$(VERSION)"
```

This `Makefile` is located under `/etc/dkms/template-dkms-mkdeb/` in my system. 
**Note:** Looks like that this file is not maintained by `dell/dkms` repo anymore. I do not know which package is installing that file.

As mentioned in the [manual](https://linux.die.net/man/8/dkms), `dkms` uses this file by default:

> It uses a template debian directory found in /etc/dkms/template-dkms-mkdeb as the basis for the package. 

But we can also define our own `Makefile` for our own module:

> Alternatively, if DKMS finds a file called /usr/src/<module>-<module-version>/<module>-dkms-mkdeb it will use that folder instead. 

### Workaround

I copied the template files to my module:

```
cp -r /etc/dkms/template-dkms-mkdeb /usr/src/<moudle>-<module-version>/<module>-dkms-mkdeb`
```

Then modified the `/usr/src/<moudle>-<module-version>/<module>-dkms-mkdeb/Makefile` like this:
```make
#/usr/bin/make
SRC = $(DESTDIR)/usr/src
SHARE = $(DESTDIR)/usr/share/$(NAME)-dkms

all:

clean:

install:

#source tree
ifeq ("$(wildcard $(NAME)-$(VERSION))", "$(NAME)-$(VERSION)")
	install -d "$(SRC)"
	cp -a $(NAME)-$(VERSION) $(SRC)
	#chmod 644 -R "$(SRC)/$(NAME)-$(VERSION)"
endif

#tarball, possibly with binaries
ifeq ("$(wildcard $(NAME)-$(VERSION).dkms.tar.gz)", "$(NAME)-$(VERSION).dkms.tar.gz")
	install -d "$(SHARE)"
	install -m 644 $(NAME)-$(VERSION).dkms.tar.gz "$(SHARE)"
endif

#postinst, only if we are supporting legacy mode
ifeq ("$(wildcard common.postinst)", "common.postinst")
	install -d "$(SHARE)"
	install -m 755 $(PREFIX)/usr/lib/dkms/common.postinst $(SHARE)/postinst
endif
``` 

Basically only commented this line:
```
#chmod 644 -R "$(SRC)/$(NAME)-$(VERSION)"
```

Since the `Makefile` is using `cp -a` to copy all the files then file permissions will be preserved. I just need to make sure my scripts have the right `x` permissions sitting at the root of my module `/usr/src/<module>-<module-version>/` :
```
-rwxr-xr-x 1 pi pi 276 Jun 12 15:42 post-install.sh
-rwxr-xr-x 1 pi pi 228 Jun 12 15:38 post-uninstall.sh
```
My `dkms.conf` has these lines:
```conf
POST_INSTALL="post-install.sh"
POST_REMOVE="post-uninstall.sh"
```

Doing these steps, `dkms mkdeb` will use the `Makefile` in my module's template to create `.deb` file properly with correct permissions and when I install `.deb` file the scripts do run fine with no error. 

Good thing about this workaround is that I don't need to touch `/etc/dkms/template-dkms-mkdeb/Makefile` and I can leave it as is, and I can just craft a custom `Makefile` for my module without affecting other modules.

### Moving Forward

My workaround may work for some other people as well, but I think the proper solution is to fix `/etc/dkms/template-dkms-mkdeb/`, so that everyone can use `dkms mkdeb` with scripts with no problem. 

Let me know if you know where `/etc/dkms/template-dkms-mkdeb/` is maintained, I will try to send a PR!

evelikov, 2023-06-15:
> Let me know if you know where /etc/dkms/template-dkms-mkdeb/ is maintained, I will try to send a PR!

@farzadpanahi as [mentioned above](https://github.com/dell/dkms/issues/53#issuecomment-955158318) distro specifics have been dropped from dkms itself. It's possible that Debian and/or Ubuntu has a copy/variant of it, but for that you'll have to check with [their gitlab instance](https://salsa.debian.org/)