dkms-project/dkms #58: Resize xterm aborts dkms install or remove with Broken pipe

severach, 2018-06-06:
Here I resize the xterm in the middle of remove and install, crashing both.

````
:: Running pre-transaction hooks...
(1/1) Remove DKMS modules
==> dkms remove perle-serial/3.9.2_4 -k 4.9.106-1-lts49
==> dkms remove perle-serial/3.9.2_4 -k 4.4.135-1-lts44
==> dkms remove perle-serial/3.9.2_4 -k 4.16.13-2-dosemu
error: command terminated by signal 13: Broken pipe
:: Processing package changes...
(1/1) reinstalling perle-serial [##] 100%
:: Running post-transaction hooks...
(1/5) sync all file systems
(2/5) Install DKMS modules
==> dkms install perle-serial/3.9.2_4 -k 4.9.106-1-lts49
==> dkms install perle-serial/3.9.2_4 -k 4.4.135-1-lts44
error: command terminated by signal 13: Broken pipe
(3/5) Reloading system manager configuration...
(4/5) Arming ConditionNeedsUpdate...
(5/5) sync all file systems
````

````
$ pacman -Q dkms gnome-terminal xterm
dkms 2.5-3
gnome-terminal 3.28.2-1
xterm 333-1
````

DKMS will complete so long as I don't resize the xterm.

This seems to be recent behavior because I've fidgeted with windows while waiting for a long time and I've only noticed this recently.

yuzaipiaofei, 2018-07-24:
@severach 
Seems like the version you use is not the latest one.
Could you test the new version to see if it can resolve your problem？

severach, 2019-02-25:
````
% pacman -Q dkms 
dkms 2.5-4
````
Seems to be fixed. Probably a problem in the Arch scripts.

closed 2019-02-25 by severach