dkms-project/dkms #59: Ensuring correct version of kernel-devel is installed

praetorian20, 2018-06-08:
Hi folks,
I have a simple kernel module that I need to distribute to CentOS 7.3 and 7.4 hosts. I've created a DKMS RPM to facilitate this and added a requirement on `kernel-devel`. 

```
Requires:	kernel-devel
```
This seemed to work at first. However, some of our 7.3 hosts have a newer kernel version installed than the stock 7.3 kernel

```bash
~% yum list installed | grep ^kernel
kernel.x86_64                        3.10.0-514.el7                  @anaconda/7
kernel.x86_64                        3.10.0-514.26.2.el7             @tbs-server
kernel-devel.x86_64                  3.10.0-514.el7                  @anaconda/7
kernel-headers.x86_64                3.10.0-514.el7                  @anaconda/7
~% uname -r
3.10.0-514.26.2.el7.x86_64
```

So `kernel-devel-3.10.0-514.26.2.el7` is required for compiling the kernel module I'm distributing, but the RPM's requirement is satisfied by the installed `kernel-devel-3.10.0-514.el7`. Due to this the `dkms build` step fails.

I cannot simply add a requirement on a specific kernel version because the 7.4 hosts have a newer version of the kernel, and this would defeat the purpose of the DKMS RPM.

How can I ensure the same `kernel-devel` package as the currently running kernel is installed? Or is there a canonical way of handling such problems? Should I check for this during RPM's `%install` and fail installation?

scaronni, 2018-06-11:
Hello, this should not happen.
If you have `kernel-3.10.0-514.26.2.el7` installed, just trying to install `kernel-devel` with `yum` should pull in the latest version of `kernel-devel`, whatever that is. Even newer than the kernel you have installed, if you have not updated in a while.

I think you can just fix this with a `yum update` or a `yum install kernel-devel-<version>` if you need an old and specific version.

praetorian20, 2018-06-11:
Hello,
`yum update` will fix the problem, but I was hoping for a solution that involves my kernel module automatically updating to the necessary kernel-devel version.

These hosts are currently in this state because the updated kernel package was pushed to them by Chef, but kernel-devel was not updated at the time. It sounds like your recommendation is to ensure the hosts have the right kernel-devel version installed by having Chef install that as well.

Is there a recommended way of handling such a situation? Should I fail installing my module? Or, in `%post` should I skip `dkms build` and `dkms install` after `dkms add`? If I do the latter, will dkms build and install my module the next time a kernel update is pushed out?

scaronni, 2018-10-08:
Sorry for the late reply. I guess that the proper fix is like you said, just make sure you don't update just parts of the kernel packages but all of the relevant subpackages as well.

closed 2018-10-08 by scaronni