dkms-project/dkms #69: --all doesn't apply commands to all kernels

martingalvan, 2018-11-23:
It seems that commands like e.g. `dkms install mydriver/1.0.0 --all` only work for kernels on which the module is already built (which would render `dkms build ... --all` useless). I would expect that `--all` would e.g. go through /lib/modules and try to build the module for all the kernels listed there. Instead, it iterates through an old kernel which I don't even have installed anymore (but had the module built back then), and the current kernel (which already has the module built).

I don't know how this would manage the kernel headers for all the different kernels, either.

yuzaipiaofei, 2018-11-29:
@martingalvan 
Could you show the dkms status before you run the "dkms install mydriver/1.0.0 --all" ?
it needs to add and build driver ,then you could install one or  you can choose  "--all "

martingalvan, 2018-11-29:
`dkms status` shows only the kernels for which the module is already built or installed. For example:

```
$ dkms status mydriver
mydriver, 1.0.0, 4.13.9-300.fc27.x86_64, x86_64: built
mydriver, 1.0.0, 4.18.16-200.fc28.x86_64, x86_64: installed
```

whereas:

```
$ ls /lib/modules
4.18.11-200.fc28.x86_64 4.18.12-200.fc28.x86_64 4.18.16-200.fc28.x86_64
```

All of those have their kernel headers installed. Doing `dkms build mydriver/1.0.0 --all` only says that the module is already built for the kernels listed by `dkms status`. I could manually issue a `dkms build` for each of the other kernels, but what's the point of `--all` then?

evelikov, 2020-01-20:
Note that the manual for "dkms install" does not list "--all" as supported command.
As such with https://github.com/dell/dkms/pull/117 the command will trigger and explicit error.

While it it possible to make it work, it is very fragile as pointed in this report.

evelikov, 2020-02-24:
With the above MR, dkms produces an error for actions which do not support it.
This enforces the documentation, which has been clear for a long time.

I think we can close this.

martingalvan, 2020-02-26:
@evelikov It hasn't been clear. `--all` is described by itself in a generic OPTIONS section on the man page, with its description implying that it can be used on various commands:

> This is useful for things like remove , mktarball , etc.

However, the only command which explicitly mentions `--all` as an option is `remove`.

scaronni, 2021-10-30:
Fixed the man page in 41efbd50f635af0ec230478525e5be56d6003896.

closed 2021-10-30 by scaronni