dkms-project/dkms #73: `dkms autoinstall` fails to substitute variables in build scripts

Skirmisher, 2019-01-03:
I have observed an inconsistency in the behavior of DKMS regarding the substitution of the variables `$kernelver` and `$arch` in build scripts.

It is documented that `$kernelver` within `dkms.conf` will be substituted with the version of the kernel the module is being built for. This appears to extend to build scripts (`PRE_BUILD`, `POST_BUILD`, etc.) specified within the conf file: guides such as [this one](https://gist.github.com/dop3j0e/2a9e2dddca982c4f679552fc1ebb18df) use it within their own build scripts, as part of a setup that presumably works for them.

I confirmed this behavior when initially performing a `dkms install` for a module on my current kernel: my script was executed with the proper values filled in for the variables. It was during a later kernel update that I noticed my modules weren't being signed automatically. After some digging, I discovered that the `dkms autoinstall` command run by the kernel update script was *not* substituting these variables in my script, naturally causing the signing to fail. (The module was otherwise built correctly.)

For now, I've rewritten my script to take these variables as command-line parameters rather than relying on substitution by DKMS. (`dkms autoinstall` still performs substitutions in `dkms.conf`; the line `POST_BUILD="sign.sh $kernelver $arch"` behaves as expected.)

I observed this behavior with DKMS version 2.6 on Fedora 29.

As a side note, I greatly appreciated the quantity of comments present in the scripts comprising DKMS itself; they were quite helpful in tracing the control flow involved, even though I never discovered the root cause. :sweat_smile:

ColMelvin, 2019-12-07:
The root cause can be found in the [`dkms`](https://github.com/dell/dkms/blob/master/dkms) binary and the problem is outlined below:

When running `dkms build …`, the script ends up calling `maybe_build_module()`, which sets the two variables (`$kernelver` & `$arch`), along with `$module` & `$module_version`, in a global scope before it [calls `build_module`](https://github.com/dell/dkms/blob/767b16c0ab580dedf31bfc39a51d61b60848f6ad/dkms#L1712) and, similarly, `maybe_install_module` sets the 4 variables before it [calls `install_module`](https://github.com/dell/dkms/blob/767b16c0ab580dedf31bfc39a51d61b60848f6ad/dkms#L1733) (which, itself, calls `build_module` as needed).  This means that all 4 variables are available to the `POST_BUILD` script.

However, when `autoinstall()` [calls `install_module`](https://github.com/dell/dkms/blob/767b16c0ab580dedf31bfc39a51d61b60848f6ad/dkms#L3387) (and the implicit `build_module`, as needed), it only sets `$module` & `$module_version` in the global scope.  For `$kernelver` & `$arch`, it relies on the already defined values, which were previously [declared in a local scope](https://github.com/dell/dkms/blob/767b16c0ab580dedf31bfc39a51d61b60848f6ad/dkms#L549-L550); as such, they are not passed on to the `POST_BUILD` script.

scaronni, 2021-08-28:
Fixed in recent merge.

closed 2021-08-28 by scaronni