dkms-project/dkms #80: should **ARCH** variable used in postinst?

drpo, 2019-05-10:
In *postinst*, **ARCH** is init to `dpkg --print-architecture`, then passed to *common.postinst*. In *common.postinst*, **ARCH** is intended to be the argument for `-a`, which is system architecture according to the manual.

So, there are two architecture, dpkg architecture and system architecture. These two are not always the same. In *common.postinst*, dpkg architecture `amd64` is translate to system architecture `x86_64`, and dpkg architecture `i?86` is translate to system architecture `i686`.

Then I doubt these two architecture can be mixing used.

In the context of *postinst*, not specific `-a` and use the default is enough. And architecture translate can be removed too.

Another reason not use **ARCH** is that it's an important variable in kernel module building.

superm1, 2019-05-11:
This does sound like it's potentially problematic in some cases to me.  Was this just an exploratory browsing that found this or did you actually have a problem?

If you had a problem could you share the details so DKMS developers can reproduce it? The case that comes to mind to me is perhaps a different architecture docker container and running DKMS there.

drpo, 2019-05-11:
Armbian and dkms postinst both use **ARCH** variable.

if you run `ARCH= dkms build` on Armbian, which is the case of dkms postinst,
you will get error `arch﻿﻿﻿﻿//Makefile﻿﻿﻿﻿﻿﻿﻿: No﻿ such﻿ file or directory﻿`

see https://forum.armbian.com/topic/10344-postinst-of-dkms-package-is-broken-on-armbian/

scaronni, 2025-01-28:
Please check the latest version, the code has been refactored many times since this bug was opened. In the unlikely event the issue is still there, please open another bug. Thanks.

closed 2025-01-28 by scaronni