dkms-project/dkms #83: DKMS Will Not Reinstall Module after Kernel Upgrade [RasPi]

Mootikins, 2019-05-23:
I have a module (a [fork](https://github.com/mootikins/mk_arcade_joystick_rpi) of [RecalBox's mk_arcade_joystick_rpi](https://github.com/recalbox/mk_arcade_joystick_rpi) for a RetroPie handheld, the [Freeplay CM3](www.freeplaytech.com)) that I have been fiddling with the the past few days, trying my hardest to get DKMS to reinstall the module whenever the kernel and kernel headers are updated, but to no avail. I have tried the following:

Editing the Makefile and dkms.conf to get the Kernel version, as demonstrated in [this bug report](https://bugs.launchpad.net/ubuntu/+source/rtl8812au/+bug/1705820]this bug report)

Various things in the AUTOINSTALL field of the dkms.conf file, all referenced in different online resources. This includes:
```
AUTOINSTALL=YES
AUTOINSTALL="YES"
AUTOINSTALL=yes
AUTOINSTALL="yes"
AUTOINSTALL=Y
```

And I have verified that I am installing the module as multiple online resources have shown:
```
sudo dkms add -m mk_arcade_joystick_rpi -v VERSION_NUMBER
sudo dkms build -m mk_arcade_joystick_rpi -v VERSION_NUMBER
sudo dkms install -m mk_arcade_joystick_rpi -v VERSION_NUMBER
#To verify installation
dkms status
#And activate
sudo modprobe mk_arcade_joystick_rpi
```

The module works as intended after the modprobe, but as soon as I update the kernel and headers with apt update and apt upgrade, the module is no longer installed. I'd like to get this figured out since it affects all the users of the handheld and so it can be implemented in other modules that we use.

jdoss, 2019-06-14:
I have the same problem with the [WireGuard kernel module](https://copr.fedorainfracloud.org/coprs/jdoss/wireguard/) on Fedora (https://lists.zx2c4.com/pipermail/wireguard/2019-February/003851.html) and I haven't been able to track it down. I beleive ZFS did too and they worked around it with this addtion to the https://github.com/zfsonlinux/zfs/blob/master/rpm/generic/zfs-dkms.spec.in#L75-L90 in their RPM. 

I haven't gotten around to trying that in the `wiredguard-dkms` package, but this feels like something DKMS itself should handle.

gregory-lee-bartholomew, 2019-11-26:
If it is a RedHat-based system, perhaps this is the issue?:

https://bugzilla.redhat.com/show_bug.cgi?id=1753044

I posted a work-around in that bug report, but I (too?) would like to see this issue properly fixed.

Mootikins, 2019-11-26:
This is Raspbian, which is (as far as I know) not running on GRUB. I will try this workaround when I get a chance.

closed 2022-07-15 by scaronni