dkms-project/dkms #84: Support for using /proc/kheaders.tar.xz

z3ntu, 2019-05-24:
Starting with Linux 5.2-rc1, the option `Enable kernel header artifacts through /proc/kheaders.tar.xz (IKHEADERS_PROC)` is available in the kernel configuration. I think it would be good to use this, if available.

The commit: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=43d8ce9d65a54846d378545770991e65838981e0

z3ntu, 2019-06-03:
And it's moved to a new position already (or soon) ;)

https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?id=f7b101d33046a837c2aa4526cef28a3c785d7af2

Now at `/sys/kernel/kheaders.tar.xz`

scaronni, 2022-05-22:
Note: https://cateee.net/lkddb/web-lkddb/IKHEADERS_PROC.html

z3ntu, 2022-05-23:
@scaronni This link is from before the commit linked in my second comment. It's now in sysfs. See https://github.com/torvalds/linux/blob/master/init/Kconfig#L707-L714 and https://cateee.net/lkddb/web-lkddb/IKHEADERS.html