dkms-project/dkms #85: dkms should offer a way to not load modules during "dkms install"

rhertzog, 2019-06-25:
In https://bugs.debian.org/931017 I reported how "dkms install" of virtualbox-guest modules breaks the running debian installer by loading the newly built modules but also other unrelated modules that were not loaded by debian-installer.

The rationale given in https://github.com/dell/dkms/commit/f5bfb12fef1fc06e56355cdba500eaa98d4e6aa8 which added this problematic modprobe call does not apply to Debian at all since the "dkms install" happens at package installation time and not on the next reboot...

So it would be nice to be able to disable this module loading feature... for the installer, where the calls to "dkms install" are embedded behind package installations, it would be best if we could set an environment variable that would disable this feature. But I imagine that an entry in the configuration file would also be useful so that the Debian package maintainer can opt to disable this entirely... a command line switch would also seem logical.

xuzhen, 2019-07-05:
sad. the pull request https://github.com/dell/dkms/pull/27 has existed for 2 years

adalton, 2019-11-26:
I have what I think is another rational for this request.

I have a containerized application that includes a kernel module that we use `dkms` to build and load. Running `dkms` to build and load our specific module has the unwanted side-effect of loading other unwanted modules. It is very counter-intuitive to call `dkms` to install a module --- by name --- and have it load other unrelated modules.

It seems like this behavior was introduced to address a use case, that it provides the desired behavior in that use case, but that the same command has other use cases where the "auto load everying" behavior is unwanted.  Exposing some option to enable users opt-in/out of that behavior would be tremendously helpful.