dkms-project/dkms #94: Be more verbose when failing due to missing module.ko

terencode, 2019-08-03:
When I was packaging a driver, it took me a long time to understand why dkms was failing and make.log didn't report any error.
It was a typo in  `BUILT_MODULE_NAME[0]` in dkms.conf that made dkms unable to locate the built module.
An incorrect Makefile configuration can produce the same result.

I suggest to append a message like the following: 
> Make sure the name of the generated module is correct and at the root of the build directory!

before  `Consult the make.log in the build directory
/var/lib/dkms/biokernbase/3.1.2.1/build/ for more information`.

eli-schwartz, 2019-08-05:
The error message could also special case this to only suggest it in situations where it is known that make exited successfully but still failed to produce the expected files.

scaronni, 2020-11-30:
https://github.com/dell/dkms/commit/5375478c996de8b14bba2cad0fde70ef573eb550

closed 2020-11-30 by scaronni