dkms-project/dkms #100: Compile using all cores (in parallel)

graysky2, 2019-10-13:
It seems that dkms uses only 1 core for building (Arch Linux using v2.7.1).  Since most systems are multicore it seems logical to build via dkms using multiple cores (akin to `make -jx`).  I found [this old Debian bug report](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=663114) that contained a patch that was indicated that an upstream merge took place.  What is the status?  Thank you.

yuzaipiaofei, 2019-11-21:
you can build your codes with make -jN to verify the building time, it can be compared to the dkms build
@graysky2

graysky2, 2019-11-21:
This is the nvidia-340xx driver on an old machine.  I will update this when I get a chance to do it.  I am unsure what the result will tell us.  When dkms builds, it only uses 1 of 4 cores (htop verified).  I am unsure what the time result will tell us that we don't already know.

GoPerry, 2020-01-10:
add those to your dkms.conf ,then you will use the dual cpu to build it.
if you have no more question about this . 
 i will close the issue ,feel free to reopen it.

# Find out how many CPU cores can be use if we pass appropriate -j option to make.
# DKMS could use all cores on multicore systems to build the kernel module.
num_cpu_cores()
{
        if [ -x /usr/bin/nproc ]; then
                nproc
        else
                echo "1"
        fi
}

MAKE[0]="make -j$(num_cpu_cores) kdir=$kernel_source_dir -C $kernel_source_dir M=$dkms_tree/$module/$module_version/build"

~                                                                                                         
~

closed 2020-01-10 by GoPerry