dkms-project/dkms #111: Missing comment to undo the "build" stage

evelikov, 2019-11-20:
Currently we have the following pair of commands

`add` - `remove`
`install` - `uninstall`

With each one recursively calling all the other respective substeps to ensure coherent and symmetric expectations.

So for example if we call:
- `uninstall` - the binary artefacts will be purged from kmod visible locations
- `remove` - will also erase the binaries produced with `build` and eradicate the dkms module metadata.

So in the former case, we still have binaries around, while in the latter we cannot re-issue `dkms build` :-\

Can we get a "unbuild" stage please? I've got a quick and dirty patch which does the job if people are interested.

evelikov, 2019-11-20:
If I understand the Debian report correctly, https://github.com/dell/dkms/pull/20 was introduced precisely with the above usecase in mind. Mind you that "fix" seemed off, so it was reverted sometimes after with https://github.com/dell/dkms/pull/77.

superm1, 2019-11-20:
@yuzaipiaofei can certainly review a patch if you want to propose one, sure.

yuzaipiaofei, 2019-11-21:
@evelikov welcome your new patch to improve the experience.

evelikov, 2020-01-16:
Thanks everyone. I've stumbled a number of bugs along the way, which I'm addressing as part of the MR. I'll send it out in the next couple of days.

evelikov, 2020-01-29:
@superm1 @GoPerry MR https://github.com/dell/dkms/pull/117 has been opened.
Any input would be appreciated.

superm1, 2020-01-29:
Once Chinese new year is over @goperry should be able to review

evelikov, 2020-02-24:
With the MR merged we can close this. Thanks everyone.

closed 2020-02-24 by evelikov