dkms-project/dkms #113: Regression when installing custom kernels on Ubuntu

cyarp, 2019-12-10:
It appears that a recent change to dkms (24a559e439f57f7e727b6d9c3395e1c11624c230) causes one of the installation scripts (/etc/kernel/postinst.d/x-grub-legacy-ec2 which calls /usr/sbin/update-grub-legacy-ec2) to fail with the following error (visible if `set -x` added to /usr/sbin/update-grub-legacy-ec2):

`RET='20 Unsupported command "*" (full line was " * dkms: running auto installation service for kernel 4.15.18custom") received from confmodule.'`

This bug was first discovered by user R T on: https://bugs.launchpad.net/ubuntu/+source/dkms/+bug/1838839.  It appears that the underlying issue, as identified by R T, is that DKMS outputs status to stdout which x-grub-legacy-ec2 attempts to interpret.

A fix that works for me is to replace the following line in kernel_postinst.d_dkms:
`exec /usr/lib/dkms/dkms_autoinstaller start $inst_kern`
with
`exec /usr/lib/dkms/dkms_autoinstaller start $inst_kern >&2`

I would submit in a pull request for this change but am not sure if it would cause a regression with other workflows.

Steps to reproduce:
Running Ubuntu 18.04.3 LTS (bionic)
1. Following the procedure at https://help.ubuntu.com/lts/installation-guide/amd64/ch08s06.html to compile a custom kernel
2. Attempt to install the kernel with dpkg

P.S.
I'm new to the linux kernel community so apologies if I am not reporting this issue in the appropriate location.

Batcastle, 2020-06-01:
I can confirm this is an issue. Drauger OS has this issue on current beta releases. Adding the fix @cyarp mentions has worked in testing, but I am getting reports that it is not working in the wild. That may be due to other issues though.

scaronni, 2022-05-25:
There has been lots of code refactoring in the last year or so, is this still happening?

closed 2022-06-09 by scaronni