dkms-project/dkms #114: "make rpm" fails;

C-Newman, 2019-12-11:
I want to install dkms itself.
Instructions say:

- Run: make rpm to `create rpm` package

The "make" tries to copy a directory specified by "../dkms" but there is no such directory by that name.  The only "dkms" is a bash script.
I must be missing something simple that will end up with me giving myself a face palm.
Here is my transcript:

\# wget https://github.com/dell/dkms/archive/v2.8.1.tar.gz
\# tar xzf v2.8.1.tar.gz
\# cd dkms-2.8.1/
\#! Note that there is only one file named "dkms" and it is a bash script:
\# file  $(find * -name dkms)
dkms: Bourne-Again shell script, ASCII text executable, with very long lines
\# 
\# make rpm
mkdir -p /lvol/newman/dkms/dkms-2.8.1/dist
tmp_dir=`mktemp -d --tmpdir dkms.XXXXXXXX` ; \
**cp -a ../dkms ${tmp_dir}/dkms-2.8** ; \
sed -e "s/\[INSERT_VERSION_HERE\]/2.8/" dkms > ${tmp_dir}/dkms-2.8/dkms ; \
sed -e "s/\[INSERT_VERSION_HERE\]/2.8/" dkms.spec > ${tmp_dir}/dkms-2.8/dkms.spec ; \
find ${tmp_dir}/dkms-2.8 -depth -name .git -type d -exec rm -rf \{\} \; ; \
find ${tmp_dir}/dkms-2.8 -depth -name dist -type d -exec rm -rf \{\} \; ; \
find ${tmp_dir}/dkms-2.8 -depth -name \*~ -type f -exec rm -f \{\} \; ; \
find ${tmp_dir}/dkms-2.8 -depth -name dkms\*.rpm -type f -exec rm -f \{\} \; ; \
find ${tmp_dir}/dkms-2.8 -depth -name dkms\*.tar.gz -type f -exec rm -f \{\} \; ; \
find ${tmp_dir}/dkms-2.8 -depth -name dkms-freshmeat.txt -type f -exec rm -f \{\} \; ; \
rm -rf ${tmp_dir}/dkms-2.8/debian ; \
sync ; sync ; sync ; \
tar cvzf /lvol/newman/dkms/dkms-2.8.1/dist/dkms-2.8.tar.gz -C ${tmp_dir} dkms-2.8; \
rm -rf ${tmp_dir} ;
**cp: cannot stat '../dkms': No such file or directory**
bash: line 2: /tmp/dkms.Sv7LDk8l/dkms-2.8/dkms: No such file or directory
bash: line 3: /tmp/dkms.Sv7LDk8l/dkms-2.8/dkms.spec: No such file or directory

yuzaipiaofei, 2019-12-18:
+ DOCDIR=/tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64/usr/share/doc/dkms-2.8
+ export DOCDIR
+ /usr/bin/mkdir -p /tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64/usr/share/doc/dkms-2.8
+ cp -pr sample.spec /tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64/usr/share/doc/dkms-2.8
+ cp -pr sample.conf /tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64/usr/share/doc/dkms-2.8
+ cp -pr AUTHORS /tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64/usr/share/doc/dkms-2.8
+ cp -pr COPYING /tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64/usr/share/doc/dkms-2.8
+ cp -pr README.md /tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64/usr/share/doc/dkms-2.8
+ exit 0
Provides: config(dkms) = 2.8-1.el7 dkms = 2.8-1.el7 dkms-minimal = 2.8
Requires(interp): /bin/sh /bin/sh /bin/sh /bin/sh
Requires(rpmlib): rpmlib(CompressedFileNames) <= 3.0.4-1 rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1
Requires(post): /bin/sh systemd
Requires(preun): /bin/sh systemd
Requires(postun): /bin/sh systemd
Requires: /bin/bash /bin/sh
Checking for unpackaged file(s): /usr/lib/rpm/check-files /tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64
Wrote: /tmp/dkms.eAkhGujY/SRPMS/dkms-2.8-1.el7.src.rpm
Wrote: /tmp/dkms.eAkhGujY/RPMS/noarch/dkms-2.8-1.el7.noarch.rpm
Executing(%clean): /bin/sh -e /var/tmp/rpm-tmp.qArhGq
+ umask 022
+ cd /tmp/dkms.eAkhGujY/BUILD
+ cd dkms-2.8
+ rm -rf /tmp/dkms.eAkhGujY/BUILDROOT/dkms-2.8-1.el7.x86_64
+ exit 0

yuzaipiaofei, 2019-12-18:
@C-Newman 
i have no this issue found on my RHEL7 system.

# git clone https://github.com/dell/dkms
#make rpm

yuzaipiaofei, 2019-12-18:
Maybe you need to check if you have the correct permission in your current build dir.
You can try to use root to build your rpm again.

GoPerry, 2019-12-22:
feel free to reopen the issue if you still have problem on your side.

closed 2019-12-22 by GoPerry