dkms-project/dkms #120: [RFE] rebuild with build --force

bill-mcgonigle, 2020-02-26:
```
# dkms build --force zfs/0.8.3
Module zfs/0.8.3 already built for kernel 4.19.0-8-amd64/4
```
Came here via https://bugs.launchpad.net/dkms/+bug/1265136

The suggestion there is to remove and build(then install) the module, but this leaves a race condition where something else could be modprobing while the sysadmin is rebuilding.  It would be sweet (and less risky) to just allow the user to rebuild in place, then install the rebuilt module.  This comes up when changing the build parameters (in my current case, enabling debug symbols on a build via conf file).

gnif, 2021-10-20:
Why this is not yet an option is beyond me... please add this dell!

gnif, 2021-11-27:
Closed because fixed? Why?

draeath, 2021-12-03:
@bill-mcgonigle why was this closed without explanation?

xuzhen, 2022-09-12:
This issue should be fixed in https://github.com/dell/dkms/commit/8dc47a275ba79df91dbd8d17f27bad4dbcd62689

closed 2022-09-12 by xuzhen