dkms-project/dkms #121: Exec format error

jessegranger, 2020-03-03:
OS: Ubuntu 19.10

Kernel: 4.15.0-1073-oem
```
> sudo apt install backport-iwlwifi-dkms
...
Building initial module for 4.15.0-1073-oem
... (no errors)
DKMS: install completed.

> sudo modprobe -v iwlwifi
insmod /lib/modules/4.15.0-1073-oem/updates/dkms/compat.ko
modprobe: ERROR: could not insert 'iwlwifi': Exec format error
```
dmesg:
```
[  816.601908] PKCS#7 signature not signed with a trusted key
[  816.601918] compat: loading out-of-tree module taints kernel.
[  816.602095] compat: module verification failed: signature and/or required key missing - tainting kernel
[  816.602145] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 0000000085c17a63, val ffffffffc0e90cd7
```

GoPerry, 2020-04-14:
@jessegranger 
Do you enable the secure boot ?
Please try to disable secure boot and reboot system to check dkms status 
#reboot
#dkms status 
Maybe you will need to re-install the package after disable secure boot.

fstaubach, 2020-04-15:
Hey,
I stumbled upon the same error and described it here on [AskUbuntu](https://askubuntu.com/questions/1225888/virtualbox-not-working-since-upgrade-to-ubuntu-19-10-from-18-04-lts-kernel-modu?noredirect=1#comment2064214_1225888) in detail.

I come more and more to the conclusion, that my error could be related to the Dell DKMS package, which is also installed on my XPS13 which causes the error as described above.
I can confirm that I have SecureBoot disabled but still running in this issue, that my virtualbox kernel module is not able to get loaded due to a wrong signature.

Any advice?

scaronni, 2022-05-25:
Instructions on module signing and secure boot are now upstream: https://github.com/dell/dkms#module-signing