dkms-project/dkms #128: Make is not a dependency of dkms package in CentOS 8

SolarGranola, 2020-05-22:
Saw this in version 2.8 on a fresh CentOS 8 image.
Make is not a dependency of the DkmS package in CentOS. This manifested as an issue for us when we started supporting CentOS 8 for our installers. 

Previously, on a clean CentOS 7 system, we could pull down the dkms package and our installer using DKMS would build. Now, on a clean CentOS 8 system, pulling down dkms is not enough. DKMS uses make to build the kernel modules, seems like it should be a dependency...

Looking at the changelog for this coderepo I noticed this: 
[image]
And make isn't listed as a dependency in CentOS 7 (using yum deplist dkms).
So maybe it's intentional but the dependency is now necessary?

scaronni, 2020-05-22:
Hi, I'm the maintainer in Fedora/EPEL, I will look into it.

scaronni, 2020-07-20:
Update is pending here, replacing the previous 2.8.2 and 2.8.3 ones: https://bodhi.fedoraproject.org/updates/FEDORA-EPEL-2020-2a4c2c399e

closed 2020-07-20 by scaronni