dkms-project/dkms #134: dkms build with --dkmstree/--sourcetree does not work anymore without root access

dbermond, 2020-07-19:
Running this command without root access now fails for me:

```
$ dkms build --dkmstree "$(pwd)" --sourcetree "$(pwd)/NVIDIA-Linux-x86_64-450.57-no-compat32" -m nvidia/450.57 -k "$(</usr/src/linux/version)"
Error! You must be root to use this command.
$ dkms --version
dkms:2.8
$ ls NVIDIA-Linux-x86_64-450.57-no-compat32/nvidia-450.57/
common  conftest.sh  dkms.conf  Kbuild  Makefile  nvidia  nvidia-drm  nvidia-modeset  nvidia-uvm
```

Using  dkms 2.8.3. This seems to be caused by commit https://github.com/dell/dkms/commit/bd13868dcb1d9d6b89cc60edcd5a7a022bce516e.

Building modules without root access was perfectly fine to up until version 2.8.1 by using the `--dkmstree` and `--sourcetree` options. The dkms [man page](https://github.com/dell/dkms/blob/v2.8.3/dkms.8#L541-L543) says that the `--sourcetree` option is "_Useful for systems that you may not have root access, but would still like to be able to build DKMS packages_", clearly stating that it should be possible to build modules without root access. This is useful, for example, for distro packagers.

Any chance that it could be possible to build modules without root access again when using `--dkmstree`/`--sourcetree`?

superm1, 2020-07-19:
By chance would you be able to send up a PR to fix it? I agree the case you described makes sense, but perhaps check_root might need to be smarter and look whether more options are set such that root isn't needed.

dbermond, 2020-07-26:
Thank you for the reply.

Yes, I can send a PR with some proposal to fix it. Thanks for asking for this.

A simple solution would be to just check for write permissions at the `$dkms_tree` path, instead of checking for root access. This would avoid the write errors as intended by the original author who introduced the root check for the `build` command. And in the circumstances that the `build` command acts like `add`, it looks like there are already root checks on the code path before copying files to `$source_tree`.

Something like this:
```diff
--- a/dkms
+++ b/dkms
@@ -3886,7 +3886,8 @@ for action_to_run in $action; do
         check_root && make_driver_disk
         ;;
     build)
-        check_root && check_all_is_banned "build" && build_modules
+        [ ! -w "$dkms_tree" ] && die 1 $"No write access for DKMS tree at ${dkms_tree}"
+        check_all_is_banned "build" && build_modules
         ;;
     add)
         check_root && check_all_is_banned "add" && add_module
```

Would this be an acceptable solution?

superm1, 2020-07-27:
Yeah I think that's a good suggestion.

PR #136 (2020-07-31): Do not require root access for the build command
Commit bd13868dcb1d9d6b89cc60edcd5a7a022bce516e broke the build command when running without root access. Building as a non-root user is useful in some cases and is described in the man page as a supported feature.

See the discussion in #134

Fixes #134
commit 94b92246 (2020-07-31): Do not require root access for the build command

Commit bd13868dcb1d9d6b89cc60edcd5a7a022bce516e broke the build
command when running without root access. Building as a non-root
user is useful in some cases and is described in the man page as
a supported feature.

See the discussion in #134

Fixes #134
closed 2020-07-31 by superm1