dkms-project/dkms #137: Modprobe Hangs

Ryushin, 2020-08-04:
I encountered issues on a server running Devuan Beowulf (Debian Buster).  It seems this command:
```find /sys/devices -name modalias -print0 | xargs -0 cat | xargs modprobe -a -b -q
```
Produces output that modprobe cannot handle and it hangs.

Output of: find /sys/devices -name modalias -print0 | xargs -0 cat
```
platform:alarmtimer
wmi:0E7AF9F2-44A1-4C6F-A4B0-A7678480DA61
acpi:PNP0C14:
platform:reg-dummy
platform:nct6775
platform:microcode
platform:intel_rapl_msr
platform:dmi-ipmi-si
acpi:PNP0C0C:
acpi:ACPI000C:
platform:GHES
platform:efi-framebuffer
platform:coretemp
platform:ipmi_bmc
platform:efivars
platform:serial8250
input:b0011v0001p0001eAB41-e0,1,4,11,14,k71,72,73,74,75,76,77,79,7A,7B,7C,7D,7E,7F,80,8C,8E,8F,9B,9C,9D,9E,9F,A3,A4,A5,A6,AC,AD,B7,B8,B9,D9,E2,ram4,l0,1,2,sfw
serio:ty06pr00id00ex00
input:b0011v0002p0006e0000-e0,1,2,k110,111,112,113,114,r0,1,6,8,amlsfw
serio:ty01pr00id00ex00
platform:i8042
input:b0010v001Fp0001e0100-e0,12,kramls1,2,fw
platform:pcspkr
platform:GHES
pci:v00008086d00000E09sv000015D9sd0000066Bbc06sc04i00
scsi:t-0x00
scsi:t-0x00
pci:v00008086d00001D02sv000015D9sd0000066Bbc01sc06i01
pci:v00008086d0000150Esv00008086sd000012A1bc02sc00i00
pci:v00008086d0000150Esv00008086sd000012A1bc02sc00i00
pci:v00008086d0000150Esv00008086sd000012A1bc02sc00i00
pci:v00008086d0000150Esv00008086sd000012A1bc02sc00i00
pci:v00008086d00001D10sv000015D9sd0000066Bbc06sc04i00
pci:v00008086d00000E24sv000015D9sd0000066Bbc08sc80i00
acpi:PNP0800:
acpi:PNP0C04:
platform:iTCO_wdt
pci:v00008086d00001D41sv000015D9sd0000066Bbc06sc01i00
acpi:IPI0001:
pci:v00008086d00000E22sv000015D9sd0000066Bbc08sc80i00
pci:v00008086d00000E02sv000015D9sd0000066Bbc06sc04i00
pci:v00008086d000010D3sv000015D9sd00000000bc02sc00i00
pci:v00008086d00001D1Esv000015D9sd0000066Bbc06sc04i00
pci:v00008086d00000E20sv000015D9sd0000066Bbc08sc80i00
pci:v00008086d00001D3Asv000015D9sd0000066Bbc07sc80i00
pci:v0000102Bd00000532sv000015D9sd0000066Bbc03sc00i00
pci:v00008086d0000244Esv000015D9sd0000066Bbc06sc04i01
pci:v00008086d00000E27sv000015D9sd0000066Bbc08sc80i00
pci:v00008086d00000E0Asv000015D9sd0000066Bbc06sc04i00
pci:v00008086d00001D22sv000015D9sd0000066Bbc0Csc05i00
pci:v00008086d00000E00sv000015D9sd0000066Bbc06sc00i00
pci:v00008086d00000E25sv000015D9sd0000066Bbc08sc80i00
usb:v1D6Bp0003d0506dc09dsc00dp03ic09isc00ip00in00
scsi:t-0x00
scsi:t-0x00
usb:v067Bp2773d0100dc00dsc00dp00ic08isc06ip50in00
usb:v1D6Bp0002d0506dc09dsc00dp01ic09isc00ip00in00
pci:v00001033d00000194sv00001033sd00000194bc0Csc03i30
pci:v00008086d00000E08sv000015D9sd0000066Bbc06sc04i00
pci:v00008086d00000E23sv000015D9sd0000066Bbc08sc80i00
usb:v8087p0024d0000dc09dsc00dp01ic09isc00ip00in00
input:b0003v0557p2221e0100-e0,1,2,3,4,k110,111,112,r8,B,a0,1,m4,lsfw
hid:b0003g0001v00000557p00002221
[... 177 lines cut ...]
modprobe: WARNING: Module platform:GHES not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module platform:efi-framebuffer not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module platform:coretemp not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module platform:ipmi_bmc not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module platform:serial8250 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module platform:i8042 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module platform:GHES not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module pci:v00008086d00000E09sv000015D9sd0000066Bbc06sc04i00 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module pci:v00008086d00001D10sv000015D9sd0000066Bbc06sc04i00 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module acpi:PNP0800: not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module acpi:PNP0C04: not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module pci:v00008086d00000E02sv000015D9sd0000066Bbc06sc04i00 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module pci:v00008086d00001D1Esv000015D9sd0000066Bbc06sc04i00 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module pci:v00008086d0000244Esv000015D9sd0000066Bbc06sc04i01 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module pci:v00008086d00000E0Asv000015D9sd0000066Bbc06sc04i00 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module pci:v00008086d00000E00sv000015D9sd0000066Bbc06sc00i00 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
modprobe: WARNING: Module pci:v00008086d00000E08sv000015D9sd0000066Bbc06sc04i00 not found in directory /lib/modules/5.6.0-0.bpo.2-amd64
insmod /lib/modules/5.6.0-0.bpo.2-amd64/kernel/sound/usb/snd-usb-audio.ko index=-2
```

Running
```insmod /lib/modules/5.6.0-0.bpo.2-amd64/kernel/sound/usb/snd-usb-audio.ko index=-2```
results in the hang.

I went down this rabbit hole when trying to update zfs dkms.  It would always hang after "depmod....." when installing and never complete.

Ryushin, 2020-08-04:
It looks like the module is the culprit.  Probably the best thing to do would be to auto kill the modprobe if it does not complete after a certain length of time.

scaronni, 2021-11-08:
Closed as with the aforementioned merge there is no automatic loading.

closed 2021-11-08 by scaronni