dkms-project/dkms #140: Fedora creates wrong symlinks for previous kernel

enigma131, 2020-09-06:
I've build Dkms in Fedora I've used before under Ubuntu for my DVB cards. Ubuntu's one have no bug.

Fedora use weak-updates for previous kernel module loading, and this contains wrong symlinks to the newest Kernel module instead the relevant kernel module.

Example, I got update to kernel 5.8. The module for kernel 5.8 works , but i have, as example for previous kernel :
ls -l /lib/modules/5.7.17-200.fc32.x86_64/weak-updates
lrwxrwxrwx. 1 root root 54 4 sept. 10:21 stb0899.ko.xz -> /lib/modules/5.8.4-200.fc32.x86_64/extra/stb0899.ko.xz
instead:
lrwxrwxrwx. 1 root root 54 4 sept. 10:21 stb0899.ko.xz -> /lib/modules/5.7.17-200.fc32.x86_64/extra/stb0899.ko.xz (module is in correct place but wrong symlink)

So when i boot to previous kernel, modules loading fails .
I know i can manually change the symlinks of each line at each kernel update, but this is not the purpose of dkms ,isen't it ?

Is it a DKMS or a Fedora issue, and how to correct it ?

enigma131, 2020-09-18:
I've found the solution for my modules under Fedora / CentOs. Thanks to the internet because no help from Fedora teams nor DKMS manual pages.
The normal way is to not use weak-updates for DKMS.
As Fedora /CentOs and I presume Redhat use weak-updates as default conf, you have to add in each dkms.conf a line :
NO_WEAK_MODULES="true" or NO_WEAK_MODULES="yes"
Tested this morning for kernel updates, ow previous kernel modules are loading

Can the man page for dkms be updated ?

scaronni, 2020-11-30:
https://github.com/dell/dkms/commit/3a1af750e88b2b1ec63280c53bbe30264197d75f

closed 2020-11-30 by scaronni

enigma131, 2020-12-09:
Finally solved, got updated dkms with updated manual