dkms-project/dkms #141: Having a hyphen '-' as part of PACKAGE_VERSION caused strange failures from the resulting deb (from launchpad issues)

hkatzdev, 2020-10-05:
Hello,
Having a hyphen in the version number will cause the package to not be installable. If you try to install the package made from `dkms mkdeb --source-only`, it will claim that it is a binaries-only package.
```
sudo dpkg -i /root/{redacted}-dkms_20201004-1_amd64.deb
Selecting previously unselected package {redacted}-dkms.
(Reading database ... 84510 files and directories currently installed.)
Preparing to unpack .../{redacted}-dkms_20201004-1_amd64.deb ...
Unpacking {redacted}-dkms (20201004-1) ...
Setting up {redacted}-dkms (20201004-1) ...
Building for 4.19.0-10-amd64 4.19.0-11-amd64
Building for architecture amd64
This package appears to be a binaries-only package
 you will not be able to build against kernel 4.19.0-10-amd64
 since the package source was not provided
This package appears to be a binaries-only package
 you will not be able to build against kernel 4.19.0-11-amd64
 since the package source was not provided
```

Adding `set -x` to `/usr/lib/dkms/common.postinst` shows that the version being passed in excludes the hyphen and everything after that, which is why it can not find it (looking in non existent directories)
```
+ [ -e /var/lib/dkms/{redacted}/20201004 ]
+ [ -f /usr/share/{redacted}-dkms/{redacted}-20201004.dkms.tar.gz ]
+ [ -d /usr/src/{redacted}-20201004 ]
```
Since the version is 20201004-1, the correct directories would be `/var/lib/dkms/{redacted}/20201004-1`, `/usr/share/{redacted}-dkms/{redacted}-20201004-1.dkms.tar.gz`, and `/usr/src/{redacted}-20201004-1`

Temporarily hardcoding the version to "20201004-1" in the above file fixed the issue and allowed the package to install.

It appears that this bug was first reported in 2010 (https://bugs.launchpad.net/dkms/+bug/599983); however, I did not see an issue here, and since this bug is still present I decided to add a Github issue.

vicamo, 2020-10-06:
In https://github.com/dell/dkms/blob/master/template-dkms-mkdeb/debian/postinst:
```
CVERSION=`dpkg-query -W -f='${Version}' $DEB_NAME | awk -F "-" '{print $1}' | cut -d\: -f2`
```
I think this might be the root cause.

However, according to [Debian Policy Manual](https://www.debian.org/doc/debian-policy/ch-controlfields.html#version):

> The upstream_version may contain only alphanumerics 6 and the characters . + - ~ (full stop, plus, hyphen, tilde) and should start with a digit. If there is no debian_revision then hyphens are not allowed.

It follows a PACKAGE_VERSION containing one hyphen here, then it would be considered an illegal debian version as it has only one hyphen in upstream version but none in the debian revision. If it contains more, then the last one should still be recognized as the hyphen in debian_revision, and there should be one removed in the upstream version. In the end, you still have no match you want.

So, I would say using hyphens as package (upstream) version when you're trying to package for debian would always bring additional troubles. Of course, it's still possible to teach DKMS to handle such cases.

hkatzdev, 2020-10-06:
Thank you for the reply. That makes sense, I didn't realize that hyphens can only be used in the upstream version if there is also a debian revision - sorry about that. Should I leave this issue open or close it since this isn't a good version number anyways?

vicamo, 2020-10-06:
I was redirected from Launchpad bug. I'd say I personally would like to take this as a chance to create test cases for mkdeb etc., but the final decision belongs to the maintainers.

scaronni, 2022-05-25:
Creation of `deb` and `rpm` packages is gone from DKMS.

closed 2022-05-25 by scaronni