dkms-project/dkms #142: DKMS not removing files - unexpected behaviour if DEST_MODULE_LOCATION doesn't start with /kernel?

elParaguayo, 2020-10-14:
I'm not sure if this is a duplicate of #64 or not as that was marked as fixed but I recently experienced this issue.

The problem I've had is that DKMS was not removing certain files when I upgraded my kernel meaning that there were some orphaned files left behind in /usr/lib/modules which needed to be manually deleted.

I posted this on the [Arch forums](https://bbs.archlinux.org/viewtopic.php?id=259647) and the issue was solved by updating the dkms.conf file. The config file had set DEST_MODULE_LOCATION to be in /extra but, according to the man page, this must be /kernel. Updating the config file so the modules were installed into /kernel fixed the problem.

Should DKMS provide a warning/error if DEST_MODULE_LOCATION if not set to the right folder?

elParaguayo, 2020-10-26:
Adding to this - it doesn't look like the "solution" above actually works.

Having just upgraded the system, I'm still left with the old kernel folder which just contains some of the files built by dkms.

Is this intended behaviour?

scaronni, 2022-05-25:
`DEST_MODULE_LOCATION` should not really matter.

The leftover files unfortunately is not the intended behaviour.

evelikov, 2023-06-06:
@elParaguayo is this still an issue? Can you provide a simple reproducer?

Something like, the following would be great:
 - install arch
 - install arch kernel package X
 - install dkms modules Y - how
 - random dkms commands
 - files Z still present, where it should be gone

elParaguayo, 2023-06-11:
Yes, it's still happening but I wonder if that's more a function of my not really understanding how DKMS works.

All of the modules that aren't deleted are listed  in `dmks.conf` but look like they're skipped during installation as it matches a module found in the kernel. However, the installation also seems to skip other modules. Should I remove these from `dkms.conf`?

evelikov, 2023-06-11:
What are these "all the modules not listed in dkms.conf"? Where do they come from? What "other modules"?

It's hard to give a meaningful answer, when there is no concrete example/reproducer. Mind you that I'm also using Arch, so following some steps won't be too burdensome for me ;-)

elParaguayo, 2023-06-11:
> It's hard to give a meaningful answer, when there is no concrete example/reproducer. Mind you that I'm also using Arch, so following some steps won't be too burdensome for me ;-)

Unfortunately, that's not quite true! I'm using this package from the AUR: https://aur.archlinux.org/packages/tbsecp3-driver-git-dkms However, you'd need to be on linux-lts as I think it's currently broken on 6.3. And, even if you are on lts 6.1, you need to amend the `source` in PKGBUILD to `'git+https://github.com/cookog/tbsecp3-driver.git#branch=6.1'`

I don't blame you if that's quite an effort!

evelikov, 2023-06-13:
> Unfortunately, that's not quite true!

I don't quite follow, what exactly is not true? I'm asking for details here.

If I cannot reproduce the issue locally and there is no _concrete example_ of a) what was done b) when happens and c) what is expected, we cannot really help you.

elParaguayo, 2023-06-13:
I was referring to your comment that it wouldn't be too difficult for you to try to replicate the issue. 

I don't want to waste your time so I'll close this for now as I probably won't have time to focus on this and it's working well enough.

Thanks for taking the time to reply.

closed 2023-06-13 by elParaguayo