dkms-project/dkms #157: build error >= 2.8.6 version of dkms

mmarcini, 2021-09-27:
Checking for unpackaged file(s): /usr/lib/rpm/check-files /tmp/dkms.DkGMVKV9/BUILDROOT/dkms-2.8.6-1.el7.x86_64
error: Installed (but unpackaged) file(s) found:
   /etc/kernel/install.d/dkms

RPM build errors:
    bogus date in %changelog: Fri Jun 17 2004 Gary Lerhaupt <gary_lerhaupt@dell.com> 1.93.04-1
    bogus date in %changelog: Tue Nov 03 2003 Gary Lerhaupt <gary_lerhaupt@dell.com> 0.44.05-1
    bogus date in %changelog: Tue May 05 2003 Gary Lerhaupt <gary_lerhaupt@dell.com> 0.28.04-1
    Installed (but unpackaged) file(s) found:
   /etc/kernel/install.d/dkms

It appears that the Makefile installs the file:
install:
        mkdir -p -m 0755 $(VAR) $(SBIN) $(MAN) $(ETC) $(BASHDIR) $(SHAREDIR) $(LIBDIR)
        mkdir -p -m 0755 $(KCONF)/install.d $(KCONF)/prerm.d $(KCONF)/postinst.d
        sed -i -e 's/#RELEASE_STRING#/$(RELEASE_STRING)/' -e 's/#RELEASE_DATE#/$(RELEASE_DATE)/' dkms dkms.8
        install -p -m 0755 dkms_common.postinst $(LIBDIR)/common.postinst
        install -p -m 0755 dkms $(SBIN)
        install -p -m 0755 dkms_autoinstaller $(LIBDIR)
        install -p -m 0644 dkms_framework.conf $(ETC)/framework.conf
        install -p -m 0755 sign_helper.sh $(ETC)
        install -p -m 0644 dkms_dbversion $(VAR)
        install -p -m 0644 dkms.bash-completion $(BASHDIR)/dkms
        gzip -c -9 dkms.8 > $(MAN)/dkms.8.gz
        chmod 0644 $(MAN)/dkms.8.gz
        **install -p -m 0755 kernel_install.d_dkms $(KCONF)/install.d/dkms**
        install -p -m 0755 kernel_postinst.d_dkms $(KCONF)/postinst.d/dkms
        install -p -m 0755 kernel_prerm.d_dkms $(KCONF)/prerm.d/dkms

But the %files directive:
%files
%defattr(-,root,root)
%doc sample.spec sample.conf COPYING README.md
%if 0%{?fedora} >= 20 || 0%{?rhel} >= 7 || 0%{?suse_version} >= 1210
%{_unitdir}/%{name}.service
%else
%{_initrddir}/%{name}_autoinstaller
%endif
%{_prefix}/lib/%{name}
%{_mandir}/*/*
%{_sbindir}/%{name}
%{_localstatedir}/lib/%{name}
%config(noreplace) %{_sysconfdir}/%{name}
# these dirs are for plugins - owned by other packages
%{_sysconfdir}/kernel/postinst.d/%{name}
%{_sysconfdir}/kernel/prerm.d/%{name}
%{_sysconfdir}/bash_completion.d/%{name}

Does not appear to have the file.

Adding the file to the repo spec file allows the build to work.

scaronni, 2021-09-28:
I see you are using the bundled SPEC file. I think the `debian` folder and the `dkms.spec` file should be removed, every distributions packages it their own way and those are pretty much abandoned.

On the same topic I'm also the Fedora/EPEL maintainer of DKMS, and dkms is already in EPEL 7 and you can install it directly from EPEL:
```
yum -y insall dkms kernel-devel
```
Or you can install directly the RPM if you are not using EPEL:
```
yum -y install kernel-devel https://kojipkgs.fedoraproject.org/packages/dkms/2.8.6/1.el7/noarch/dkms-2.8.6-1.el7.noarch.rpm
```

mmarcini, 2021-09-28:
> I see you are using the bundled SPEC file. I think the `debian` folder and the `dkms.spec` file should be removed, every distributions packages it their own way and those are pretty much abandoned.
> 

Ok.   So what should the README.md say with respect to this?

Adding the %files line clearly works.

> On the same topic I'm also the Fedora/EPEL maintainer of DKMS, and dkms is already in EPEL 7 and you can install it directly from EPEL:
> 

We know EPEL works...

> ```
> yum -y insall dkms kernel-devel
> ```

I agree, epel is the way to go for RHEL 7.x amd 8.x and we will probably do that, but I haven't found an equivalent for SLES 12.x and 15.x that doesn't involve messing with entitlements during OS to OS changeover across our fleet of servers.

Do you happen to know about SLES equivalents to EPEL?

scaronni, 2021-09-28:
Also, can you try using the SPEC file here to build it? https://src.fedoraproject.org/rpms/dkms/raw/rawhide/f/dkms.spec
Or directly rebuild the `src.rpm`, for example: https://koji.fedoraproject.org/koji/buildinfo?buildID=1828050

scaronni, 2021-09-28:
Also, I see here, all the old Suse stuff in the repository is outdated (I see Suse 9 and 10 references, which are long EOL): https://en.wikipedia.org/wiki/SUSE_Linux#SUSE_distributions

scaronni, 2021-10-08:
Distro-specific packaging is being dropped. Almost all targets (deb creation, debian support, RPM, kmp, kmod etc.) are severely outdated.