dkms-project/dkms #162: Rework module signing

scaronni, 2021-10-08:
Configuration should be as follows:
- Create private key and public key in `/etc/dkms`
- Enable configuration to use the keys in `/etc/dkms/framework.conf`
- Modules are signed at every rebuild.

Configuration:
- Entries in `/etc/dkms/framework.conf` for keys
- File `/etc/dkms/mok.key` points to the Machine Owner Key private key
- File `/etc/dkms/mok.der` points to the Machine Owner Key certificate in DER format

Main command:
- No more old specifics Ubuntu commands in the `sign_build` function
- Merge of the `sign_helper.sh` script in the `sign_build` function
- Drop `sign_tool` variable and documentaton
- Make sure paths for the `sign-file` binary are taken into account for the various distributions

Docs:
- Add man page entry on the new configuration.
- Update README.md with information and screenshots for MOK enrollment

scaronni, 2022-05-22:
@Tachi107 I've updated the description for the certificate location, this should cover your requirements.

scaronni, 2022-05-22:
HI @Tachi107, please have a look at: https://github.com/dell/dkms/tree/rework_signing

It's a work in progress, but I think covers all use cases. You can have your DKSM keys managed the "Debian way" if you want to but at the same time covers any other use cases. The complexity is also pretty low.

scaronni, 2022-05-22:
Sample output:
```
# dkms build nvidia/515.43.04
Sign command: /lib/modules/5.17.7-300.fc36.x86_64/build/scripts/sign-file
Signing key: /etc/dkms/mok.key
Public certificate (MOK): /etc/dkms/mok.pub
Creating symlink /var/lib/dkms/nvidia/515.43.04/source -> /usr/src/nvidia-515.43.04

Building module:
Cleaning build area...
'make' -j2 module SYSSRC=/lib/modules/5.17.7-300.fc36.x86_64/build IGNORE_XEN_PRESENCE=1 IGNORE_PREEMPT_RT_PRESENCE=1 IGNORE_CC_MISMATCH=1...........................
Signing module /var/lib/dkms/nvidia/515.43.04/build/nvidia.ko
Signing module /var/lib/dkms/nvidia/515.43.04/build/nvidia-modeset.ko
Signing module /var/lib/dkms/nvidia/515.43.04/build/nvidia-drm.ko
Signing module /var/lib/dkms/nvidia/515.43.04/build/nvidia-uvm.ko
Signing module /var/lib/dkms/nvidia/515.43.04/build/nvidia-peermem.ko
cleaning build area...
```

Tachi107, 2022-05-22:
I've only took a really brief look at this so I can't say much, but isn't storing keys in /etc a bit odd? What about putting them under `/var/lib/dkms`?

scaronni, 2022-05-22:
To be honest I find it odd the other way around, `/var/` is for variable state stuff

A certificate used to sign that you enroll in the firmware is a configuration item, something that you can preconfigure and push to multiple systems either (signed with your own company CA, etc.). `akmods` does the same, it's in `/etc`.

scaronni, 2022-05-22:
I've added another commit that checks for Secure Boot and enrollment only if Secure Boot is enabled. So you can still sign modules if SecureBoot is not enabled.

scaronni, 2022-05-22:
Btw, this all comes from real use cases, In my company we push our machine certificates for network access generated from our own CA, these certificates have a shorter life time than the 36500 days in DKMS and we replace them regularly. With the same certificates we are signing the modules regardless of SecureBoot and importing them into the TPM if SecureBoot is enabled.

So far we've done the same but with the previous logic.

Tachi107, 2022-05-22:
> To be honest I find it odd the other way around, `/var/` is for variable state stuff
>
> A certificate used to sign that you enroll in the firmware is a configuration item, something that you can preconfigure and push to multiple systems either (signed with your own company CA, etc.).

From the FHS:

- [`/var/lib`](https://refspecs.linuxfoundation.org/FHS_3.0/fhs-3.0.html#varlibVariableStateInformation): _"This hierarchy holds state information pertaining to an application or the system. State information is data that programs modify while they run, and that pertains to one specific host."_
- [`/etc`](https://refspecs.linuxfoundation.org/FHS_3.0/fhs-3.0.html#etcHostspecificSystemConfiguration): _"The `/etc` hierarchy contains configuration files. A "configuration file" is a local file used to control the operation of a program"_

If dkms itself creates something during its execution, it should not write to `/etc`, `/etc` should only be modified by the system administrator (programs should never write to `/etc` at all, as it may be read-only in hardened environments). In my opinion the correct approach here would be to put automatically generated keys in `/var/lib/dkms`, but make them overridable by the system administrator by putting custom keys in `/etc/dkms`. Similar to how systemd [handles unit](https://www.freedesktop.org/software/systemd/man/systemd.unit.html) files; vendor units are under `/usr/lib/systemd`, runtime units are in `/run/systemd`,  but they can be both overridden locally by placing units in `/etc/systemd`

> I've added another commit that checks for Secure Boot and enrollment only if Secure Boot is enabled. So you can still sign modules if SecureBoot is not enabled.

Nice :)

> Btw, this all comes from real use cases, In my company we push our machine certificates for network access generated from our own CA, these certificates have a shorter life time than the 36500 days [...]

The approach I described earlier is compatible with this use case, as the keys put in `/etc` by the system admin would be automatically read by dkms.

scaronni, 2022-05-22:
You convinced me, they are now in `/var/lib/dkms`.

I've rebased the branch, can you make a test on your system? You can just call `make` and then copy over the resulting `dkms` script in your path somewhere where it overrides your system one.

```
# dkms build xpadneo/0.9.1
Sign command: /lib/modules/5.17.7-300.fc36.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...
Creating symlink /var/lib/dkms/xpadneo/0.9.1/source -> /usr/src/xpadneo-0.9.1

Building module:
Cleaning build area...
make -j8 KERNELRELEASE=5.17.7-300.fc36.x86_64 -C /lib/modules/5.17.7-300.fc36.x86_64/build M=/var/lib/dkms/xpadneo/0.9.1/build VERSION=v0.9.1 modules...
Signing module /var/lib/dkms/xpadneo/0.9.1/build/hid-xpadneo.ko
Cleaning build area...
```

scaronni, 2022-05-22:
If you are on Debian/Ubuntu would be nice to see if you can use the Debian provided tools just fine by setting 
```
mok_signing_key="/var/lib/shim-signed/mok/MOK.der"
mok_certificate="/var/lib/shim-signed/mok/MOK.priv"
```
in `/etc/dkms/framework.conf`.

scaronni, 2022-05-22:
I've updated the documentation as well: https://github.com/dell/dkms/tree/rework_signing#module-signing

scaronni, 2022-05-22:
Pushed to master with also the man page update. Will wait a few days for feedback and additional changes, thanks.

Tachi107, 2022-05-22:
Sorry for my lack of action, but I've been busy with a couple of things. I'll test your changes in the coming days

scaronni, 2022-05-24:
Sure, no worries. Thanks.

closed 2022-06-16 by scaronni

Tachi107, 2022-06-17:
Sorry, I almost forgot about this! I'm still quite busy with exams, but sooner or later you'll receive some feedback :)

scaronni, 2022-06-18:
ok thanks, feel free to create MR if you see anything not working properly. Thanks.

Safari77, 2022-06-19:
So when upgrading to dkms 3.0.4, my modules would not be signed correctly and the system would not boot.
I have `CONFIG_MODULE_SIG_SHA256=y`, dkms hardcodes algo to sha512.
I have also `CONFIG_MODULE_SIG_KEY="certs/signing_key.pem"`.

I had `sign_tool=/etc/dkms/sign-sha256.sh` with contents of
```
#!/usr/bin/sh
/lib/modules/"$1"/build/scripts/sign-file sha256 /lib/modules/"$1"/build/certs/signing_key.{pem,x509} "$2"
```
now dkms creates a new certs and keys with the same config?

Safari77, 2022-06-21:
I use dkms only for zfs modules with 5.15 LTS kernel on Fedora.
UEFI boot does not work for some reason so SecureBoot cannot be used.
For me changing to sha512 is not a problem, but it is 2.5 times slower than sha256.

I'll let you know if just changing mok_signing_key and mok_certificate does not work the next time I build a kernel. Already changed to use sha512.

AfroThundr3007730, 2022-06-26:
Well it was certainly fun tracking down why my system failed to boot because a dkms module required for boot was signed with the wrong certificate.

Perhaps instead of removing the old variable and breaking existing setups, just mark as deprecated but still use it where defined. This would result in fewer surprises.