dkms-project/dkms #167: Uninstall fail to properly remove modules when manually signed

sprshp, 2021-10-16:
I'm using secure boot and i'm signing my modules at each module/kernel update.
My signed modules are stored in the path: /lib/modules/[kernel version]/updates/dkms/

The way I'm doing is leading to an issue when uninstalling a kernel:
Because of the signature, the modules binaries located in /lib/modules/[kernel version]/updates/dkms/ differs from the version of /var/lib/dkms/zfs/kernel-[kernel version]-[arch]/module.
Therefore the function find_actual_dest_module_location locate the modules properly, but the comparison of binaries fails and the correct path is not returned:

The issue is localed in the /usr/sbin/dkms script, in the below function:
find_actual_dest_module_location()
{
...
    _for l in $locations; do
        installed=$(compressed_or_uncompressed "${install_tree}/${kernelver}${l}" "${dest_module_name[${count}]}")
        if [[ -n "${installed}" ]] && **diff "${dkms_owned}" "${installed}"** > /dev/null 2>&1; then
            echo "${l}"
            return 0
        fi
    done_
...
}

it end-up with this kind of message:
_zavl.ko:
 - Uninstallation
   - Deleting from: **/lib/modules/5.10.0-8-amd64/**
**rmdir: impossible de supprimer '': Aucun fichier ou dossier de ce type**
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version._

And of course the path /lib/modules still contains unexpected old kernel versions directories.

sprshp, 2021-10-16:
Manually signing modules seems to be a bad practice for dkms behavior.
Properly setting the automatic signature on build solve the issue.

scaronni, 2021-11-04:
So this should be closed? You should sign as described in the README (for now).

closed 2021-11-07 by sprshp