dkms-project/dkms #170: Error when building driver with 2.8.8

gitdwarf, 2021-10-30:
when trying to build a driver with dkms, the following error appears

Error! REMAKE_INITRD has been deprecated.
An error occurred. dkms add error = 5

(and the driver is not built)

This error does not occur with dkms-2.8.7, only with 2.8.8

scaronni, 2021-11-01:
Pulling in @evelikov .

Can you please paste the DKMS file for the specific module?

gitdwarf, 2021-11-01:
PACKAGE_NAME="rtl8821au"
PACKAGE_VERSION="5.12.5.2"
BUILT_MODULE_NAME[0]="8821au"
DEST_MODULE_LOCATION[0]="/kernel/drivers/net/wireless"
MAKE="'make' -j$(nproc) KVER=${kernelver} KSRC=/lib/modules/${kernelver}/build"
AUTOINSTALL="yes"
REMAKE_INITRD="no"

evelikov, 2021-11-01:
@gitdwarf simple fix is to remove the `REMAKE_INITRD` line all together. The `no` option is the default so there will be no difference in behaviour.

- Although I can change things to silently continue, if the value is "no". Maybe even provide more meaningful warning/error message.

- I was thinking that we ought to bump the dkms version to `3.0.0` since a lot of cruft was removed - this way it's dead obvious.

@scaronni let me know what you think about each of the above.

scaronni, 2021-11-03:
> @gitdwarf simple fix is to remove the `REMAKE_INITRD` line all together. The `no` option is the default so there will be no difference in behaviour.
> 
> * Although I can change things to silently continue, if the value is "no". Maybe even provide more meaningful warning/error message.

I think we should print that is deprecated and move on. Or we could also document how a distribution package maintainer should configure `/etc/dkms/framework.conf` for the distribution specific "hooks", like rebuilding intirds etc.

> * I was thinking that we ought to bump the dkms version to `3.0.0` since a lot of cruft was removed - this way it's dead obvious.

Fine for me.

> Edit: seems like 2.8.8 was released, yet `dkms --version` says `dkms-2.8.7`. Forgot to update the version perhaps?

Yes: https://github.com/dell/dkms/blob/master/Makefile#L4

scaronni, 2021-11-03:
> Edit: seems like 2.8.8 was released, yet `dkms --version` says `dkms-2.8.7`. Forgot to update the version perhaps?

I will add a `make release` target to the Makefile that changes version, date and creates tag if you don't beat me at it.

scaronni, 2021-11-04:
Fixed version (3.0.0) and warnings on deprecated options instead of error in `master`. @evelikov feel free to improve on it.

closed 2021-11-04 by scaronni