dkms-project/dkms #176: where is --no-clean-kernel --no-prepare-kernel    options  ?

limbo127, 2021-11-16:
Hello, 
since dkms 3.0.1 no more options, how can we do to have the same comportement
Regards 
Nicolas

scaronni, 2021-11-16:
Hello, can you paste a link or something about your DKMS file? Thanks.

limbo127, 2021-11-16:
hello,
this is from a arch linux standard installation   of virtualbox module
before yesterday ( with dkms arch) this command pass without issue, now
after upgrade dkms to 3.0.1 , not

yes |pacman -Syyu  virtualbox-host-dkms    # install dkms packages
# some var
VBOX=$(pacman -Q --info  virtualbox-host-dkms -q -e |grep Version|awk
'{print $3}'|sed 's/-.*//g')
VBOXHOST="${VBOX}_OSE"
# rebuild
dkms build -m vboxhost -v $VBOXHOST -k $KVER  --kernelsourcedir=$KDIR
--no-clean-kernel --no-prepare-kernel

scaronni, 2021-11-16:
Those are deprecated, you can just remove them. I will add them here so you will get a warning: https://github.com/dell/dkms/commit/fff3032e5d471e3a1f456ec8fa1b20002024eb30

scaronni, 2021-11-16:
https://github.com/dell/dkms/commit/3eb09af9358a1bd55b9ca04866b5161e714bd5f2.patch

closed 2021-11-17 by scaronni