dkms-project/dkms #177: <module>/<module-version> is not optional

netboy3, 2021-11-16:
The combination of 90a7002c300d068df231b576724c7effa2803569 and e73f9bbf20130163d3f47b788cb41dde9865a4b1 cause a mandatory requirement of module/module-version to both add and install. This change in logic makes it impossible to run a very common DKMS scenario such as `dkms install .` inside a DKMS module source directory.

fishilico, 2021-11-16:
Hello,
Since DKMS 3.0.0, it has also no longer been possible to run `dkms add /path/to/source-tree`. For example chipsec is hitting this issue in its continuous integration system (https://github.com/chipsec/chipsec/pull/1305#issuecomment-970693380):

```console
$ dkms add drivers/linux
Error! Arguments <module> and <module-version> are not specified.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>
```

But according to the source code of `add_module`, this usage of specifying a source directory with neither `-m` or `-v` is supposed to be supported: https://github.com/dell/dkms/blob/3eb09af9358a1bd55b9ca04866b5161e714bd5f2/dkms.in#L788-L789

Is this dead code? What is the new way to add or install modules from custom paths?

By the way, I worked around my issue by patching `dkms`:

```diff
diff --git a/dkms.in b/dkms.in
index 4d54378b9b44..d959dbb75e88 100644
--- a/dkms.in
+++ b/dkms.in
@@ -2347,7 +2347,7 @@ remove | unbuild | uninstall)
     ${action}_module
     ;;
 add | build | install)
-    check_module_args $action
+    [[ $action = add ]] || check_module_args $action
     check_all_is_banned $action # TODO: fix/enable --all
     [[ $action = install ]] && check_root || check_rw_dkms_tree
     ${action}_module
```

Should I send this patch in a Pull Request?

kakra, 2021-11-17:
> But according to the source code of `add_module`, this usage of specifying a source directory with neither `-m` or `-v` is supposed to be supported:

I can confirm that it broke exactly that assumption for our project.

scaronni, 2021-11-17:
> Should I send this patch in a Pull Request?

Yes please, thanks!

Unfortunately until a new release is out, nobody tests what code has been pushed to the repository; so nobody reports bugs before the release. But nothing prevents to make quick release with fixes.

kakra, 2021-11-17:
> nobody tests what code has been pushed to the repository

Having tests as suggested in #173 would mostly fix that. It would also make it easier to submit a test for a failing operation as part of a bug report.

closed 2021-11-17 by scaronni