dkms-project/dkms #187: why drop mkdeb or mkrpm?

keepingxt, 2021-12-09:
I found the mkdeb/rpm function have been droped in the latest version? And my project is using the feature. Is there a other recommended way to make deb/rpm pakage? Thanks.

scaronni, 2021-12-09:
Hello, anything distribution specific regarding packaging was working out of pure luck as it was all severely outdated.

I'm maintaining a few DKMS RPM SPEC files which cover the `mkrpm` part, basically:

- https://github.com/negativo17/dkms-evdi
- https://github.com/negativo17/dkms-nvidia
- https://github.com/negativo17/dkms-xpadneo

How did you use it? Can you show anything?

keepingxt, 2021-12-10:
Hi, Scaronni thanks for replay and share. 
I  used `dkms mkrpm` based on `template-dkms-mkrpm.spec` to make rpm pakages at centos 7.4. 
So I need use `rpmbuild` directly to make rpm pakages after `mkrpm` removed from dkms. I don't know if I understand correctly. Thanks.

scaronni, 2021-12-15:
If you provide me a link with what you're trying to achieve i can probably provide you the SPEC files to use.

closed 2022-05-22 by scaronni

atJoe, 2022-12-17:
I was happily using mkrpm as well.  I was using it for a source only distribuition.  Now to get an old install with v2.8.7 so I can have it back.  I get it that there are more and more distros every day and that they all can't live in this project.  It would be nice if there was a way for us to fork off this capability this off as a dkms-rpm project that would depend on the base dkms project.  I'm imagining that lovers of other distros could make their own dkms-packagetype projects if they want this kind of functionality.  I'll take a look at the spec files you've posted and the one generated by v2.8.7 to see what the path looks like for hand-creating what mkrpm used to do for me.

scaronni, 2022-12-17:
Contribution specific for distributions has always been basically zero, and every piece of code, from building packages to manipulating ram disk was basically abandoned. It's much better if the person who work with DKMS inside the distributions would provide the relevant functionality somewhere else.

In the case of RPMS, here are some examples I maintain:

- https://github.com/negativo17/dkms-xone
- https://github.com/negativo17/dkms-xpadneo
- https://github.com/negativo17/dkms-nvidia
- https://github.com/negativo17/dkms-evdi

As you can see from all of these, it basically boils down to these in the SPEC file:
```
%install
# Create empty tree:
mkdir -p %{buildroot}%{_usrsrc}/%{dkms_name}-%{version}/
cp -fr module/* %{buildroot}%{_usrsrc}/%{dkms_name}-%{version}/

%if 0%{?fedora}
# Do not enable weak modules support in Fedora (no kABI):
install -p -m 644 -D %{SOURCE2} %{buildroot}%{_sysconfdir}/dkms/%{dkms_name}.conf
%endif

%post
dkms add -m %{dkms_name} -v %{version} -q || :
# Rebuild and make available for the currently running kernel:
dkms build -m %{dkms_name} -v %{version} -q || :
dkms install -m %{dkms_name} -v %{version} -q --force || :

%preun
# Remove all versions from DKMS registry:
dkms remove -m %{dkms_name} -v %{version} -q --all || :
```

That's it, you don't need anything else.

stewpend0us, 2024-12-18:
Since I have to package my own debs now what is your advice for my postinst script?

* should I be calling
```dkms install -m <name> -v <version>```
in this case if the install fails it leaves my (apt) installation in a sketchy state that's annoying to clean up
* or should I be calling
```/usr/lib/dkms/common.postinst```
is common.postinst legacy? or is it going to be maintained?
what are the proper arguments for this?
will I have the same issue with a failed install or does it make that better?