dkms-project/dkms #219: Skip building modules for incompatible kernel configs

xnox, 2022-03-24:
In Ubuntu, we ship various dkms modules and we also have many different kernel types.

Some dkms modules target hardware & kernels that require certain configuration options to be enabled in the kernel. Most of Ubuntu kernels have those options turned on, but not others.

At the same time when developing kernels we test a cross combination of all dkms modules against all kernel modules.

It would be nice to have something like a `BUILD_EXCLUSIVE_KERNEL=` option but something along the lines of `BUILD_REQUIRED_CONFIG=` such that one can do something like

`BUILD_REQUIRED_CONFIG=CONFIG_ACPI`

And if a given kernel doesn't have ACPI turned on, dkms module build will not be attempted and instead it would be skipped, declaring that this dkms module is incompatible with a given kernel.

Ditto other scenarios similar to that one - i.e. when dkms module plugs into a wider kernel framework that might be turned off in a given kernel. For example dahdi-linux dkms unable to be built against kernels that have CONFIG_AUDIO turned off.

Or can you please suggest any other options as to how dkms.conf can include some kernel config validation checks (either declarative or scripts) which are not based on kernel version number; but based on kernel config features.

xnox, 2022-03-25:
I ended up doing:

```
if ! grep -q '^CONFIG_ECHO=[ym]$' $kernel_source_dir/.config || ! grep -q '^CONFIG_FW_LOADER=[ym]$' $kernel_source_dir/.config; then
BUILD_EXCLUSIVE_KERNEL="^$"
fi
```

Not sure if this is the best way.

anbe42, 2022-06-03:
There is also the other way around, i.e. a certain configuration must *not* be enabled. E.g. the zfs and nvidia modules cannot be built on PREEMPT_RT kernels (due to transitive usage of GPL-only symbols).

Could perhaps be modeled as
BUILD_REQUIRED_CONFIG="!CONFIG_PREEMPT_RT"

In general, this should allow a space (or comma) separated list of (negated) options.

anbe42, 2022-06-13:
A preliminary patch that adds a (yet undocumented) `BUILD_EXCLUSIVE_CONFIG` option can be found here:
https://gist.github.com/anbe42/66aacc5a4ee1c91089cf11f02d677e3f

scaronni, 2022-08-08:
Can you make a pull request with the updated option also in the man page? thanks.

evelikov, 2022-10-21:
@anbe42 seems like this landed in Debian. As mentioned above - can you make this a proper PR and while also updating the manual page?

anbe42, 2022-10-26:
See pull request #269.

commit da958b29 (2023-04-20): add BUILD_EXCLUSIVE_CONFIG

support build exclusion depending on kernel features being present/absent,
e.g. BUILD_EXCLUSIVE_CONFIG="CONFIG_FOO !CONFIG_BAR"

Closes #219
closed 2023-04-20 by evelikov