dkms-project/dkms #222: Nvidia module fails against Clang-built kernel

RyanHakurei, 2022-05-07:
CC: @nickdesaulniers 

This seems like the same error encountered when trying to build modules against Clang pre DKMS 3.0.2.

```
==> dkms install --no-depmod nvidia/510.68.02 -k 5.17.5-256-tkg-cfs-llvm
Error! Bad return status for module build on kernel: 5.17.5-256-tkg-cfs-llvm (x86_64)
Consult /var/lib/dkms/nvidia/510.68.02/build/make.log for more information.
```

```
DKMS make.log for nvidia-510.68.02 for kernel 5.17.5-256-tkg-cfs-llvm (x86_64)
Sat May  7 07:01:21 PM EDT 2022
make[1]: Entering directory '/usr/lib/modules/5.17.5-256-tkg-cfs-llvm/build'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:729: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/lib/modules/5.17.5-256-tkg-cfs-llvm/build'
make: *** [Makefile:82: modules] Error 2
```

flindeberg, 2022-05-08:
Which version / commit of `DKMS` are you using?  And you have headers present as well? (i.e. `linux517-tkg-cfs-llvm-headers` unless my TKG-fu is off).

RyanHakurei, 2022-05-08:
DKMS is sitting at 3.0.3, and I do have the headers installed as well (the PKGBUILD automatically builds the header package in addition to the kernel).

nickdesaulniers, 2022-05-08:
What is `5.17.5-256-tkg-cfs-llvm`?

flindeberg, 2022-05-08:
> What is `5.17.5-256-tkg-cfs-llvm`?

It's just a kernel produced by a well used collection of build scripts. See https://github.com/Frogging-Family/linux-tkg for more info. `cfs` for the default Linux-scheduler (**C**ompletely **F**air **S**cheduler), llvm for llvm / clang. 

I re-built a clean version of a vanilla linux-tkg-kernel with llvm and it seems to work for me. I cannot reproduce the error in this issue.

@RyanHakurei Can you share your build configuration for linux-tkg?

RyanHakurei, 2022-05-09:
It's worth noting that I am patching in the TaskType CPU Scheduler, hence why it would appear by the package name that I am using cfs. This does build just fine though and DKMS will build against the kernel when using gcc.

```
# linux-TkG config file

# Linux distribution you are using, options are "Arch", "Void", "Ubuntu", "Debian", "Fedora", "Suse", "Gentoo", "Generic".
# It is automatically set to "Arch" when using PKGBUILD.
# If left empty, the script will prompt
_distro="Arch"

# Kernel Version - Options are "5.4", "5.7", "5.8", "5.9", "5.10", "5.11", "5.12", "5.13", "5.14", "5.15", "5.16", "5.17"
_version=""

#### MISC OPTIONS ####

# External config file to use - If the given file exists in path, it will override default config (customization.cfg) - Default is ~/.config/frogminer/linux-tkg.cfg
_EXT_CONFIG_PATH=~/.config/frogminer/linux-tkg.cfg

# [Arch specific] Set to anything else than "true" to limit cleanup operations and keep source and files generated during compilation.
# Default is "true".
_NUKR="true"

# [install.sh specific] Git mirror to use to get the kernel sources, possible values are "googlesource.com" and "kernel.org"
_git_mirror="kernel.org"

# Custom compiler root dirs - Leave empty to use system compilers
# Example: CUSTOM_GCC_PATH="/home/frog/PKGBUILDS/mostlyportable-gcc/gcc-mostlyportable-9.2.0"
CUSTOM_GCC_PATH=""

# Custom LLVM compiler root dirs - Leave empty to use system llvm compiler
# Example: CUSTOM_LLVM_PATH="/home/frog/PKGBUILDS/mostlyportable-llvm/llvm-mostlyportable-11.0.0"
CUSTOM_LLVM_PATH=""

# Set to true to bypass makepkg.conf and use all available threads for compilation. False will respect your makepkg.conf options.
_force_all_threads="true"

# Set to true to prevent ccache from being used and set CONFIG_GCC_PLUGINS=y (which needs to be disabled for ccache to work properly)
_noccache="false"

# Set to true to use modprobed db to clean config from unneeded modules. Speeds up compilation considerably. Requires root - https://wiki.archlinux.org/index.php/Modprobed-db
# Using this option can trigger user prompts if the config doesn't go smoothly.
# !!!! Make sure to have a well populated db !!!! - Leave empty to be asked about it at build time
_modprobeddb="false"

# modprobed-db database file location
_modprobeddb_db_path=~/.config/modprobed.db

# Set to "1" to call make menuconfig, "2" to call make nconfig, "3" to call make xconfig, before building the kernel. Set to false to disable and skip the prompt.
_menunconfig="false"

# Set to true to generate a kernel config fragment from your changes in menuconfig/nconfig. Set to false to disable and skip the prompt.
_diffconfig=""

# Set to the file name where the generated config fragment should be written to. Only used if _diffconfig is active.
_diffconfig_name=""

# [install.sh specific] Use tmpfs as a work directory, recommended when RAM >= 32GB to reduce HDD/SSD usage. For more information, see https://wiki.archlinux.org/title/Tmpfs
_use_tmpfs="false"

# [install.sh specific] tmpfs folder path, only used when _use_tmpfs="true".
#                       Creates a linux-tkg work folder within that pathmake sure to have nothing important in "$_tmpfs_path/linux-tkg"
_tmpfs_path="/tmp"

# [install.sh: Generic and Gentoo specific] Dracut options when generating initramfs
_dracut_options="--lz4"

#### KERNEL OPTIONS ####

# Name of the default config file to use for the kernel
# Default (empty)  : "config.x86_64" from the linux-tkg-config/5.y folder.
[... 176 lines cut ...]
#### USER PATCHES ####

# community patches - add patches (separated by a space) of your choice by name from the community-patches dir
# example: _community_patches="clear_nack_in_tend_isr.myrevert ffb_regression_fix.mypatch 0008-drm-amd-powerplay-force-the-trim-of-the-mclk-dpm-levels-if-OD-is-enabled.mypatch"
_community_patches=""

# You can use your own patches by putting them in a subfolder called linux<version>-tkg-userpatches (e.g. linux510-tkg-userpatches) next to the PKGBUILD and giving them the .mypatch extension.
# You can also revert patches by putting them in that same folder and giving them the .myrevert extension.

# Also, userpatches variable below must be set to true for the above to work.
_user_patches="true"

# Apply all user patches without confirmation - !!! NOT RECOMMENDED !!!
_user_patches_no_confirm="false"

#### CONFIG FRAGMENTS ####

# You can use your own kernel config fragments by putting them in the same folder as the PKGBUILD and giving them the .myfrag extension.

# Also, the config fragments variable below must be set to true for the above to work.
_config_fragments="true"

# Apply all config fragments without confirmation - !!! NOT RECOMMENDED !!!
_config_fragments_no_confirm="false"
```

evelikov, 2022-05-09:
If it were me, I would first ensure the combination works _regardless_ of dkms or `5.17.5-256-tkg-cfs-llvm` - by doing the out-of-tree module build by hand, in a vanilla kernel tree.

Without looking too closely, it seem like this is not a dkms problem.

flindeberg, 2023-05-06:
Did you manage to find the issue here? I remember trying to reproduce it like a year ago and failing to reproduce it.

scaronni, 2025-01-15:
Should be fixed since 3.1.2.

closed 2025-01-15 by scaronni