dkms-project/dkms #229: Module signing on Arch Linux

inglor, 2022-07-02:
Module signing in Arch Linux causes error:
```
/usr/bin/dkms: line 1033: sha512: command not found
```

This is introduced from commit 3323033dfdc5f58afc303736f6b580bf62442ef7 and the case in line 1028 assumes the `sha512sum` executable is `sha512`. 

Specifically for Arch Linux the executable is `/usr/bin/sha512sum`

[3323033dfdc5f58afc303736f6b580bf62442ef7]: https://github.com/dell/dkms/commit/3323033dfdc5f58afc303736f6b580bf62442ef7

maxyz, 2022-07-02:
Actually the problem is that the variable $sign_file is not being set, same commit at the start of prepare_signing function it tries to detect the location of the file for "known" distributions.

I think something like:
```diff
diff --git a/dkms.in b/dkms.in
index ba312c3..d47c343 100644
--- a/dkms.in
+++ b/dkms.in
@@ -866,11 +866,12 @@ prepare_signing()
ubuntu* )
sign_file="/usr/src/linux-headers-$kernelver/scripts/sign-file"
;;
-            fedora* | rhel* | ovm*)
-                sign_file="/lib/modules/$kernelver/build/scripts/sign-file"
-                ;;
* )
-                echo "Binary sign-file not found, module won't be signed"
+                if [ -f "/lib/modules/$kernelver/build/scripts/sign-file" ]; then
+                    sign_file="/lib/modules/$kernelver/build/scripts/sign-file"
+                else
+                    echo "Binary sign-file not found, module won't be signed"
+                fi
;;
esac
fi
@@ -912,6 +913,11 @@ prepare_signing()
;;
*)

+            if [ ! "${sign_file}" ]; then
+                do_signing=0
+                return
+            fi
+
if [ ! -f $mok_signing_key ] || [ ! -f $mok_certificate ]; then
echo "Certificate or key are missing, generating self signed certificate for MOK..."
openssl req -new -x509 -nodes -days 36500 -subj "/CN=DKMS module signing key" \
```
would fix this.

But it would be better to get ride of all the distribution specific parts.

ideal, 2022-07-03:
I had the same problem, but sign-file exists:

```bash
$ file /usr/lib/modules/5.18.8-arch1-1/build/scripts/sign-file
/usr/lib/modules/5.18.8-arch1-1//build/scripts/sign-file: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=a8a4bcbc4ea7a3220c18418e6fe2165df6872b97, for GNU/Linux 4.4.0, with debug_info, not stripped
```

Seems that old dkms did not find the correct sign-file path.

DerVerruckteFuchs, 2022-07-03:
I had the same issue, but made the mistake of rebooting afterward. That resulted in an unbootable system since my boot partition was stuck with the previous kernel initramfs. All necessary kernel modules for mounting any filesystems and for any networking were missing due to the kernel mismatch. Perplexingly, my zfs root filesystem mounted fine in Arch's recovery mode. I could not mount my vfat boot partition or any ext4 partitions.

I applied the patch from #230 by editing the `/usr/bin/dkms` file by hand. Some chroot time with an Arch install ISO and re-installation of kernels and headers afterward fixed it for me.

Long story short, the #230 patch works for me on my machine.

simona70, 2022-07-03:
I must not restart the system? What I must do?

jansohn, 2022-07-03:
> I must not restart the system? What I must do?

Downgrade dkms and re-install kernel and headers.

    sudo pacman -U /var/cache/pacman/pkg/dkms-3.0.3-1-any.pkg.tar.zst
    sudo pacman -Sy linux linux-headers

smac89, 2022-07-04:
Now might be as good a time as ever to get acquainted with [`kernel-modules-hook`](https://archlinux.org/packages/community/any/kernel-modules-hook/). Enjoy worry-free upgrades for your kernel.

**NOTE**: I should add that this will only work if the upgrade process _completed_. If  you stop it halfway through the hook won't be able to run the [`PostTransaction`](https://archlinux.org/pacman/alpm-hooks.5.html#_actions) hooks which restores your old kernel.

isaacnugroho, 2022-07-04:
I had the same issue, I created /usr/bin/sha512 soft linked to /usr/bin/sha512sum to solve.
`sudo ln -s /usr/bin/sha512sum /usr/bin/sha512`
and then reinstall linux

ideal, 2022-07-04:
> Perplexingly, my zfs root filesystem mounted fine in Arch's recovery mode. I could not mount my vfat boot partition or any ext4 partitions.

Did you mean fallback mode by 'recovery mode'? But fallback mode uses same kernel as default mode, so I am wondering why that works.

SjonHortensius, 2022-07-04:
> But it would be better to get ride of all the distribution specific parts.

yep this feels like browser-detection all over again. Feature detection is much better

Phundrak, 2022-07-04:
@isaacnugroho I’m going to quote Jonathon on the [Archlinux bug tracker](https://bugs.archlinux.org/task/75212)
> Changing `sha512` to `sha512sum` is incorrect, `sha512` is an option being passed to the `kmodsign` (line 1030) or `$sign_file` (line 1033) command.

isaacnugroho, 2022-07-04:
@Phundrak thank you. I'll use the patch instead.

justinkb, 2022-07-04:
> @Phundrak, the comment is for the previous comment, replacing sha512 with sed: `sed -i 's/sha512/sha512sum/g' "$pkgname.in"`

my dude, you are simply not understanding the problem

justinkb, 2022-07-04:
I'm not sure this actually needs addressing other than writing about it in the arch wiki. Isn't it just the arch user's responsibility to configure sign_file and mok_signing_key and mok_certificate in /etc/dkms/framework.conf?

debohman, 2022-07-04:
This is correct. Arch needs to set the following in the dkms config:

```
sign_file='/usr/lib/modules/${kernelver}/build/scripts/sign-file'
```

The single quotes are necessary because $kernelver is not yet set when the configuration files are sourced.

> I'm not sure this actually needs addressing other than writing about it in the arch wiki. Isn't it just the arch user's responsibility to configure sign_file and mok_signing_key and mok_certificate in /etc/dkms/framework.conf?
> 
> edit: CONFIG_MODULE_SIG_FORCE is not set for arch standard kernel, so as a workaround most arch users should just set sign_file to "/usr/bin/true" or something for now.

thaewrapt, 2022-07-05:
> I'm not sure this actually needs addressing other than writing about it in the arch wiki. Isn't it just the arch user's responsibility to configure sign_file and mok_signing_key and mok_certificate in /etc/dkms/framework.conf?

You are correct in the part where `framework.conf` should be fixed for Arch but it's still a bug if `dkms` tries to run an empty command for whatever reason (and both former and latter should be addressed indeed).

W2AlharbiMe, 2022-07-05:
> > I must not restart the system? What I must do?
> 
> Downgrade dkms and re-install kernel and headers.
> 
> ```
> sudo pacman -U /var/cache/pacman/pkg/dkms-3.0.3-1-any.pkg.tar.zst
> sudo pacman -Sy linux linux-headers
> ```

@jansohn thank you so much this fixed my issue 

---

also i have something to report here after i got the issue the system refused to restart from kde application launcher widget
in case anyone got the same issue this is what i did `systemctl reboot`

gardotd426, 2022-07-06:
>You are correct in the part where framework.conf should be fixed for Arch but it's still a bug if dkms tries to run an empty command for whatever reason (and both former and latter should be addressed indeed).

Absolutely, Arch users themselves should not have to edit framework.conf to be able to have a functioning system. I've gone ahead and done it to fix the issue until something else is done (and it does fix it), but the issue doesn't just lie with Arch, I'd say it's still a bug in DKMS.

debohman, 2022-07-06:
> Absolutely, Arch users themselves should not have to edit framework.conf to be able to have a functioning system. I've gone ahead and done it to fix the issue until something else is done (and it [...]

I don't think that anybody is proposing that people will need to fix the configuration for themselves. That part of the problem can be resolved by a drop in configuration change when the project is built and packaged for Arch.

I agree that there also seems to be an issue here which needs to be examined further.

WhyNotHugo, 2022-07-08:
Given that this fails because a variable in unset (and that variable is the **command**), shouldn't `set -u` be used here, to hard-fail in situations like this?

Thomasasia, 2022-07-14:
How do I fix framework.conf?

aghpub, 2022-07-14:
On Arch? I created a file (custom.conf) in `/etc/dkms/framework.conf.d `and added the following line to it

    sign_file='/usr/lib/modules/${kernelver}/build/scripts/sign-file'

That fixed the problem for me.

closed 2022-07-15 by scaronni

Ouack23, 2022-08-30:
After https://github.com/dell/dkms/commit/a1d32c9f4fb65567dd929305b5ab5af3c5b969cf i had to leave `sign_file` commented out and put `${kernelver}` in the `mok` variables. So much time lost ...

PR #230 (2022-07-15): Look for sign-file regardless
Fix #229
commit a1d32c9f (2022-07-15): Look for sign-file regardless

Fix #229