dkms-project/dkms #236: Syntax not supported in old bash version: bad substitution

recolic, 2022-07-14:
At <https://github.com/dell/dkms/blob/1e24a54acaa2e3a8aa6e8e24b21f47553370c4fe/dkms.in#L864>, it's failing on bash 5.1.16(1). It seems that this syntax is not supported by this bash version. 

```
==> dkms install --no-depmod nvidia/515.57 -k 5.18.11-hardened1-1-hardened
/usr/bin/dkms: line 864: /usr/lib/linux-kbuild-${$kernelver%.*}/scripts/sign-file: bad substitution
==> WARNING: `dkms install --no-depmod nvidia/515.57 -k 5.18.11-hardened1-1-hardened' exited 1
```

## System Info

ArchLinux, bash 5.1.016-1, dkms-3.0.4, kernel 5.18.11-hardened1-1-hardened

scaronni, 2022-07-15:
Fixed in #230.

closed 2022-07-15 by scaronni

recolic, 2022-07-15:
Hi. This PR is not fixed by PR #230 .  It's a bash command substitution issue, and it's not related to Arch Linux support. 

I could still reproduce this issue in dkms 3.0.6.  

[image]

recolic, 2022-07-15:
I will suggest a PR to solve this issue soon.

PR #237 (2022-07-15): Bad substitution bugfix: resolve #236
Bad substitution bugfix: resolve #236