dkms-project/dkms #244: Module signing is broken in latest dkms on debian

blastrock, 2022-09-09:
Hi,

I've been using dkms signing capabilities with sign_tool and the sign_helper script, but the last update of my debian unstable broke it all.

```
Sign command: /usr/lib/linux-kbuild-5.19/scripts/sign-file
Signing key: /root/dkms.key
Public certificate (MOK): /root/dkms.der
Binary update-secureboot-policy not found, modules won't be signed
```

There seem to be multiple issues with the script, or maybe I'm not using it like I'm supposed to, please tell me.

1. There is a typo here https://github.com/dell/dkms/blob/49d647b6476d605feb10548e33fc3807415ec724/dkms.in#L893 . We check for kmodsign and then print kmod-sign in the error message.
2. These two conditions look reversed to me https://github.com/dell/dkms/blob/49d647b6476d605feb10548e33fc3807415ec724/dkms.in#L892-L900 . They should be `if [[ ! -x ...`
3. We expect debian to have kmodsign https://github.com/dell/dkms/blob/49d647b6476d605feb10548e33fc3807415ec724/dkms.in#L890 , but I definitely can't find it on my debian, isn't that a ubuntu thing? There is this issue opened https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=939392 about that.
4. And finally, I don't get why we create a new key and enroll it on every run here https://github.com/dell/dkms/blob/49d647b6476d605feb10548e33fc3807415ec724/dkms.in#L905-L906 . The keys I sign with are already enrolled, why would that be necessary?

xuzhen, 2022-09-09:
2\. This bug was fixed by https://github.com/dell/dkms/pull/242

4\. update-secureboot-policy won't create new key if /var/lib/shim-signed/mok/MOK.der exists, and won't reenroll it.
```
'--new-key')
    create_mok
    exit 0
    ;;

'--enroll-key')
    if [ -e "$SB_KEY" ]; then
        if mokutil --test-key "$SB_KEY" | \
                grep -qc 'is not'; then
            do_enroll=1
        fi
    else
        echo "No MOK found."
        exit 1
    fi
    ;;
```
```
create_mok()
{
    if [ -e "$SB_KEY" ]; then
        return
    fi

    echo "Generating a new Secure Boot signing key:"
    openssl req -config /usr/lib/shim/mok/openssl.cnf \
        -subj "/CN=`hostname -s | cut -b1-31` Secure Boot Module Signature key" \
        -new -x509 -newkey rsa:2048 \
        -nodes -days 36500 -outform DER \
        -keyout "$SB_PRIV" \
        -out "$SB_KEY"
}
```
```
SB_KEY="/var/lib/shim-signed/mok/MOK.der"
```
But we don't need update-secureboot-policy if a custom key file other than /var/lib/shim-signed/mok/MOK.der was used.

xuzhen, 2022-09-09:
I rewrote the signing parts for Ubuntu/Debian. https://github.com/xuzhen/dkms/commit/bc6853cc57f4ef589cb5cef7030c22da5eea6b94

The ```update-secureboot-policy --enroll-key ``` was removed, because you can do nothing with the debconf dialog brought up by update-secureboot-policy, and there is no enrolling action for other distributions.

I tested it on Ubuntu, @blastrock, could you test it on Debian? https://github.com/xuzhen/dkms/tree/ubuntu

blastrock, 2022-09-10:
Tested it. It works, as in it signs my modules.

I didn't test the MOK thing since I use my own keys that I enrolled myself in UEFI setup.

However, there seem to be some other errors:

```
╰─➤  sudo ./dkms remove virtualbox/6.1.38 -k 5.19.0-1-amd64 
Module virtualbox-6.1.38 for kernel 5.19.0-1-amd64 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

vboxdrv.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.19.0-1-amd64/
rmdir: failed to remove '': No such file or directory
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

vboxnetadp.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.19.0-1-amd64/
rmdir: failed to remove '': No such file or directory
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

vboxnetflt.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.19.0-1-amd64/
rmdir: failed to remove '': No such file or directory
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod...
egrep: warning: egrep is obsolescent; using grep -E

╰─➤  sudo ./dkms install virtualbox/6.1.38 -k 5.19.0-1-amd64
Sign command: /usr/lib/linux-kbuild-5.19/scripts/sign-file
Signing key: /root/dkms.key
Public certificate (MOK): /root/dkms.der

Building module:
Cleaning build area...
make -j8 KERNELRELEASE=5.19.0-1-amd64 -C /lib/modules/5.19.0-1-amd64/build M=/var/lib/dkms/virtualbox/6.1.38/build.....
Signing module /var/lib/dkms/virtualbox/6.1.38/build/vboxdrv/vboxdrv.ko
Signing module /var/lib/dkms/virtualbox/6.1.38/build/vboxnetadp/vboxnetadp.ko
Signing module /var/lib/dkms/virtualbox/6.1.38/build/vboxnetflt/vboxnetflt.ko
Cleaning build area...

vboxdrv.ko:
Running module version sanity check.
Error! Module version 6.1.38_Debian for vboxdrv.ko
is not newer than what is already found in kernel 5.19.0-1-amd64 (6.1.38_Debian).
You may override by specifying --force.

vboxnetadp.ko:
Running module version sanity check.
Error! Module version 6.1.38_Debian for vboxnetadp.ko
is not newer than what is already found in kernel 5.19.0-1-amd64 (6.1.38_Debian).
You may override by specifying --force.

vboxnetflt.ko:
Running module version sanity check.
Error! Module version 6.1.38_Debian for vboxnetflt.ko
is not newer than what is already found in kernel 5.19.0-1-amd64 (6.1.38_Debian).
You may override by specifying --force.
depmod...
```

Those rmdir and module version errors do not seem to be regressions though as they occur with debian's dkms 3.0.6 too.

xuzhen, 2022-09-12:
The rmdir error was fixed with https://github.com/dell/dkms/commit/6fab150cb4daf15d556b7e5860d64b834736f903

xuzhen, 2022-09-12:
The module version errors should be discussed in a separate issue. I close this one.

closed 2022-09-12 by xuzhen